vlachoudis/bCNC #1055: Autoleveling issue

WimRoose, 2018-11-20:
Hi

I've spent some time already finetuning soft- and hardware setup for PCB milling with flatcam + bCNC. After a lot of trial and error, I'm still struggling trying to achieve a consistent cutting depth. I've solved several issues causing this (bad clamping of pcb, broken V-bit tips, cutting depth too high, ...). My latest project however, a 150/100 mm pcb, again faces the issue of traces not being cut deep enough. Here's a [picture](https://drive.google.com/open?id=1K3aX7FxyTDpU2VLrJGBWQjNDkP0KxmuE). Note that some traces on the right side are already milled three times, so that explains a deeper cutting depth on the picture. But one can see the gradially reduced cutting depth going from bottom left to right side.

Autolevel probe is OK, I can see a gradially increase from +/- 0 (all left), to 1.9 (all right). [Here's my probe file](https://drive.google.com/open?id=1pFoJ0u2Dz4Tdiec8K3Ak9yLorXpkV9xg).

I'm using woodpecker gbrl 0.9j, latest bCNC (release). Cut depth: -1.0

I'm following wiki instructions to probe & autolevel: x=0, y=0 (bottom left), probe Z, set zero, also zero in autolevel menu.

In the gcode editor, selecting a trace on the right (so with an Z offset of about 1.8), I can see that Z command shows -1.0, so not applying autoleveling probe. Is that normal? So autoleveling correction is done on-the-fly by bCNC? I'm out of ideas right now..

Harvie, 2018-11-20:
> So autoleveling correction is done on-the-fly by bCNC?

Yes

This is strange. There's one more guy complaining about similar proble. Please take a look if the issue sounds familiar to you #771

WimRoose, 2018-11-20:
Yes & no, I already went through all autolevel issues filed. I've already countered 2 possible causes:

* pre-tension on the board: I changed the way I clamp the pcb to the bed (see picture). There's no more play on the surface off the PCB

* No play on Z-axis. After the job was finished, I homed and did a new probe on Z-axis to check if a piece of the bit tip was broken off, or whether any play on Z-axis had occured. But the difference was neglectable: something around 0.03 mm. After re-probing, I selected only some of the traces on the right (bad side), and restarted the job, but without any noticable result.

Trying to narrow down to something: is it possible to export gcode WITH autolevel correction, so I can try one single trace in the bad area with UGS or something?

Thanks for helping!

Harvie, 2018-11-20:
Here you can  apply the autolevel data on selected g-code. But remember that bCNC will still autolevel it on the fly second time. So be sure to zero/unload the autolevel data from bCNC afterwards if you are going to cut such g-code using bCNC.

[image]

WimRoose, 2018-11-20:
Okay, thanks. Will try that tonight. I've noticed that applying autolevel values to the original code removes the initial cut depth.. E.g. given a probe value of 0.5, autolevel applied value becomes +/- -0.5, but should be -1.5. Or am I mistaken here?

colesnicov, 2018-11-28:
Try `Tools > Stock > "Surface Z" `set to 0.0

FrenkK, 2019-02-21:
Like Harvie said: just perform a scan and run the gcode, the autolevel operation happens on the fly. If you click Autolevel to correct the gcode, the correction gets applied twice. I lost a lot of time finding this out ...