vlachoudis/bCNC #1057: Proposed UI changes

Harvie, 2018-11-21:
I have some proposals for UI change. It is based on my personal experience with bCNC.
And it also reflects the needs of people who are running bCNC on embedded devices. Like the guy on #1033 He will not be able to use bCNC if we'll not fix this before next release.

I've tried to implement all of these UI changes and it took me hours, but it always turned out that i don't know how tkinter works. @vlachoudis Maybe you could be a help with this one, since you are a specialist on bCNC UI :-)

Here is brief drawing of what i think should be done to improve usability:

[image]

leofaoro, 2018-11-23:
I like your proposed changes, and I have a couple of comments.
- For the Config area, perhaps it’s more relevant to go into the File tab, as config/preferences typically go there (at least in Macs). Plus it is not a frequently used area once you get bCNC working.
- I think you should have the cycle start/pause/stop buttons always present regardless of what tab you’re in. At the very least the Stop button. As I have not installed physical buttons yet for my setup only the e-stop is setup), I have struggled more than once to change tabs to find the stop button, so having that easily reachable is good for safety.

Thanks
Leo

Harvie, 2018-11-23:
>  cycle start/pause/stop buttons always present regardless of what tab you’re in

YES!

Harvie, 2018-11-23:
Also i would like to have OPTION to show editor in CAM tab, since lots of times i need to use it to select the right block, when working with CAM features.

[image]

leofaoro, 2018-11-23:
> Maybe this button should go to "control" and rest to the "file".

My philosophy for this would be to have everything related to “setttings/options” bundled (as it is now). You probably don’t want to have to go to different places to change “global” settings.

MARIOBASZ, 2018-11-29:
Hello, I just read this thread.
The button I use it sometimes, very punctually if I want to modify the height of Z surface or Z safe

vlachoudis, 2019-01-08:
@Harvie the new update contains "almost" all of the changes. Moving the editor on the right is possible but requires more tweaking.
"Stock" is used by the default CAM operations (Cut, Profile, Drill...) when no information is provided on the material, thickness, safe z etc is taken from the stock

Harvie, 2019-01-08:
Perfect! Thank you very much!

Harvie, 2019-01-09:
Problem is that none of these buttons really work now:
[image]

in fact they do work, but they open dialog in CAM tab, which is not displayed at the moment you click it.
as temporary workaround i can suggest adding some call that would switch the tabs when clicked.

vlachoudis, 2019-01-09:
@Harvie that why it was there in the first place  :)

Harvie, 2019-01-09:
@vlachoudis makes sense :-D it would be cool if we were able to display and hide these panels independently on tab selected in ribbon. That would also enable us to view the g-code editor in CAM tab.

MARIOBASZ, 2019-02-01:
Duplicating islands in editor is useful to me

Harvie, 2019-02-26:
Maybe having the left panel with separate tabs would help. But i don't know. Probably it would be too confusing like this:

[image]

But the tabs can be only virtual, so it would be possible to switch them based on what menu item people choose...

Harvie, 2019-03-21:
Here is code to switch ribbon pages, i think it can be used:

```python
                # Restore last page
                self.pages["Probe"].tabChange() # Select "Probe:Probe" tab to show the dialogs!
                self.ribbon.changePage(Utils.getStr(Utils.__prg__,"page", "File"))

                probe = Page.frames["Probe:Probe"]
```