vlachoudis/bCNC #1060: Calibrate not working

leofaoro, 2018-11-22:
I am trying to set up manual tool changes, but calibrate is not working for me. I am using today’s Git version running under Windows 10.
I home my machine, set the WCS. I have also set the change and probe coordinates. When I try to calibrate the tool goes to the probe location, feeds down and stops on contact with the probe, but the number never gets entered into the interface.
Tks

Harvie, 2018-11-22:
Have you ever used this calibration succesfully? What was the last version when this was working?
I've never really understood how to use this. I guess it just needs better documentation or more streamlined UI. Something is in the wiki.

leofaoro, 2018-11-22:
No, I have not gotten this to work, as it works in tandem with tool change. To my understanding this feature works like this: you set your workspace Z0. Then you calibrate your tool length, and the value shown will be the difference between your Z0 (WCS) and the calibration Z (i.e. the calibration touch plate). From this on every tool change it goes to the calibration tool plate to measure the difference between the old tool and the new tool.

Thanks
Leo

Harvie, 2018-11-22:
Is there somebody who knows how is this supposed to work? @Effer @vlachoudis