vlachoudis/bCNC #1061: Tool change commands not being sent

leofaoro, 2018-11-22:
Using 0.9.14 from today’s Git. When I try to send M6 commands (using the manual tool change option), nothing happens. Even manually entering M6 T1 nothing happens. I tried to change the setting to ignore, close, then reopen bCNC, still nothing.
Tks

Harvie, 2018-11-22:
@leofaoro have you ever succesfully executed toolchange using bCNC on your machine? what was last working version? unfortunately i've never tried this feature and i don't really know how to use it, so it might be bit tricky to debug this.

leofaoro, 2018-11-22:
I haven’t run it successfully but using 0.9.11 and selecting the manual tool change it would at least to to the tool change location and pause. With this version it just quits running the g-code.

Harvie, 2018-11-22:
Can you run bCNC in console (cmd) and show me the output. Are there any errors shown in cmd window when it unexpectedly quits the operation?

leofaoro, 2018-11-23:
Sure. This is what happens when I run bCNC.bat, connect to grbl, then home the machine, then execute a tool change. Just to be clear the application does not terminate, it stays open and functional. The status bar 

```
new-config Utils <ConfigParser.ConfigParser instance at 0x0000000003953948>
Controller state changed to: Alarm
Controller state changed to: Home
Controller state changed to: Idle
Job done. Purging the controller

```
And this is terminal output within bCNC

```
Grbl 1.1g ['$' for help]
$H
[MSG:'$H'|'$X' to unlock]
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$g
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
m5
ok
Run ended
2018-11-22 20:55:19.830000
Current: 3 [29]  Completed: 6% [2s Tot: 20s Rem: 18s]
Grbl 1.1g ['$' for help]
```

Also looking at the status bar it appears that the controller goes through several instructions, but really nothing happens.

thanks!!

leofaoro, 2018-11-23:
It appears that once WAIT is called, nothing is sent to Grbl.

Harvie, 2018-11-23:
Try now!

Harvie, 2018-11-23:
I've just tested and toolchange seems to be working

Harvie, 2018-11-23:
I've also fixed #1062, so toolchange workflow should be much smoother now.
Please let me know if this works for you, so i can close this issue.