vlachoudis/bCNC #1063: I can not install the latest version

1f7, 2018-11-23:
I try install to Ubuntu 16.04 according to the last instruction with no luck
```bash
orangepi@OrangePI:~$ pip2 install --upgrade bCNC  
Collecting bCNC  
  Using cached https://files.pythonhosted.org/packages/fe/b2/96469773dfba552ea45dcbf2cacc255d5d1d651f6cfe8b0a75beb082d2cc/bCNC-0.9.14.39.tar.gz  
Collecting pyserial<=3.0.1 (from bCNC)  
  Using cached https://files.pythonhosted.org/packages/42/68/41bab8a5e3e50705f25358f043181242fa197f313aa0476908d9852c8dc4/pyserial-3.0.1.tar.gz  
Collecting numpy>=1.15.4 (from bCNC)  
  Using cached https://files.pythonhosted.org/packages/2d/80/1809de155bad674b494248bcfca0e49eb4c5d8bee58f26fe7a0dd45029e2/numpy-1.15.4.zip  
Collecting opencv-python>=3.4.2.17 (from bCNC)  
  Could not find a version that satisfies the requirement opencv-python>=3.4.2.17 (from bCNC) (from versions: )  
No matching distribution found for opencv-python>=3.4.2.17 (from bCNC)  
You are using pip version 8.1.1, however version 18.1 is available.  
You should consider upgrading via the 'pip install --upgrade pip' command.
```
Previous versions were installed without problems.
opencv-python need compile by hand?

1f7, 2018-11-24:
> Does `sudo apt-get install python-opencv` help?
```bash
 sudo apt-get install python-opencv
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-opencv is already the newest version (2.4.9.1+dfsg-1.5ubuntu1.1).
The following packages were automatically installed and are no longer required:
  libllvm3.8 libmircommon5
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

> Does `sudo pip2 install opencv-python` work?
```bash
 sudo pip2 install opencv-python
The directory '/home/orangepi/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/orangepi/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting opencv-python
  Could not find a version that satisfies the requirement opencv-python (from versions: )
No matching distribution found for opencv-python
You are using pip version 8.1.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

orangepi@OrangePI:~/Downloads$ pip2 install opencv-python
Collecting opencv-python
  Could not find a version that satisfies the requirement opencv-python (from versions: )
No matching distribution found for opencv-python
You are using pip version 8.1.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```
`Python 2.7.12`

Harvie, 2018-11-24:
From OpenCV project page: https://pypi.org/project/opencv-python/#description

> **Q: Pip fails with `Could not find a version that satisfies the requirement` ...?**
> 
> A: Most likely the issue is related to too old pip and can be fixed by running pip install --upgrade pip. Note that the wheel (especially manylinux) format does not currently support properly ARM architecture so there are no packages for ARM based platforms in PyPI. However, opencv-python packages for Raspberry Pi can be found from https://www.piwheels.org/.

Harvie, 2018-11-24:
I've uploaded new version bCNC==0.9.14.41 to pypi, which should install with opencv version 2.4+
Try that pls.

Harvie, 2018-11-24:
Otherwise you can install opencv manualy.
Here is the list of packages: https://www.piwheels.org/simple/opencv-python/
You can install it directly like this:

```
pip2 install https://www.piwheels.org/simple/opencv-python/opencv_python-3.4.3.18-cp35-cp35m-linux_armv7l.whl#sha256=201422601fc99c89556c47538cb6f423f0cc1e974684701b9431bfa132fcac13
```

Harvie, 2018-11-24:
Please let me know how this turned out... Especialy what exactly solved the problem and if other packages installed well. I will add that to wiki.

1f7, 2018-11-24:
i delete cache and try also with error
```bash
orangepi@OrangePI:~$ pip2 install --upgrade bCNC
Collecting bCNC
  Downloading https://files.pythonhosted.org/packages/08/58/96d0d4781b863804a63cd1aca05a718c0245b15a8cf1e6a09d33b872c465/bCNC-0.9.14.41.tar.gz (1.1MB)
    100% |████████████████████████████████| 1.1MB 147kB/s
Collecting pyserial<=3.0.1 (from bCNC)
  Downloading https://files.pythonhosted.org/packages/42/68/41bab8a5e3e50705f25358f043181242fa197f313aa0476908d9852c8dc4/pyserial-3.0.1.tar.gz (134kB)
    100% |████████████████████████████████| 143kB 513kB/s
Collecting numpy>=1.15.4 (from bCNC)
  Downloading https://files.pythonhosted.org/packages/2d/80/1809de155bad674b494248bcfca0e49eb4c5d8bee58f26fe7a0dd45029e2/numpy-1.15.4.zip (4.5MB)
    100% |████████████████████████████████| 4.5MB 51kB/s
Collecting opencv-python>=2.4 (from bCNC)
  Could not find a version that satisfies the requirement opencv-python>=2.4 (from bCNC) (from versions: )
No matching distribution found for opencv-python>=2.4 (from bCNC)
You are using pip version 8.1.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

second try
```bash
orangepi@OrangePI:~$ pip2 install https://www.piwheels.org/simple/opencv-python/opencv_python-3.4.3.18-cp35-cp35m-linux_armv7l.whl#sha256=201422601fc99c89556c47538cb6f423f0cc1e974684701b9431bfa132fcac13
opencv_python-3.4.3.18-cp35-cp35m-linux_armv7l.whl is not a supported wheel on this platform.
You are using pip version 8.1.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```
platform is Ubuntu 16.04

 > OrangePI is ARMv7
  
Yes x32

In first post i wrote - both version work fine  with git cloned repo.
pip installer not work

Harvie, 2018-11-24:
> opencv_python-3.4.3.18-cp35-cp35m-linux_armv7l.whl is not a supported wheel on this platform.

Sorry, i've just found it's only fro python 3.4 and 3.5, so it will not work with bCNC

> In first post i wrote - both version work fine with git cloned repo.
> pip installer not work

Yes, but that is because you have already installed some packages. I need this to work using pip install, otherwise there will be

PannDemic, 2018-12-04:
So I am on a raspberry pi and I receive the same error, opencv-python>=2.4 
I have python-opencv version 2.4.9.1 installed. 
Sorry that I'm not the most knowledgeable about RPi or Raspbian. What information do you need to assist in trouble shooting?

Harvie, 2018-12-04:
> What information do you need to assist in trouble shooting?

I've referenced the pypa issues. This is not really problem of bCNC. It has to be solved in the pypi/pip.
But if they don't fix this by the time we will be releasing 0.9.15 (the first stable release of bCNC that will officialy support pip install) we can use the workaround.

For your convenience i've removed opencv from requirements, so pip install will work until they fix this.

Harvie, 2018-12-14:
Since this is blocked by upstream, but workaround was applied, i've removed this from 0.9.15 milestone.
Now it should install on ARM, but webcam support is not installed by default (even on non-ARM systems).
Once pip will be able to handle soft-dependencies i will re-add opencv, so it will work on all platforms.

Mahmouderr404, 2019-04-16:
Hellp my pleas

pip install upgrade
Collecting upgrade
  Could not find a version that satisfies the requirement upgrade (from versions: )
No matching distrib

Harvie, 2019-04-16:
@Mahmouderr404 please tell us more. What OS/distro? What python version?... please read wiki and README file.

Harvie, 2019-07-19:
@1f7 @PannDemic Can you please check that everything works now on ARM?

In #1180 i've added condition, which should prevent opencv from being required on ARM, but i don't have raspi, so can't really check that...

alien18331, 2019-09-24:
Hello
I have jetson nano(Cortex-A57 ARMv8-A 64-bit) with ubuntu 18.04.3 LTS
Can BCNC working on this? Thanks you

I install failed & get below massage
```
ERROR: Could not find a version that satisfies the requirement opencv-python>=2.4 (from bCNC) (from versions: none)
ERROR: No matching distribution found for opencv-python>=2.4 (from bCNC)
```
My installation flow is below

```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install git python python-tk python-pip
sudo apt-get install python-pip python-setuptools python-dev python-tk libjpeg-dev zlib1g zlib1g-dev
sudo pip install --upgrade pip
sudo apt-get install python-opencv   
sudo pip2 install --upgrade bCNC
```
----------------------------------------------------------
 <name> --version
```
pip&pip3 : pip 19.2.3 from /home/smg/envs/TF2_ECP_EBR/lib/python3.6/site-packages/pip (python 3.6)
pip2 : pip 19.2.3 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)
```

Harvie, 2019-09-24:
setup.py defines following:
`'opencv-python>=2.4 ; "arm" not in platform_machine'`

which means opencv-python should not be installed on arm based systems at all, because it's not available at the moment. But your machine still tries to do so.

Can you please somehow extract `platform_machine` string from your python installation and send it to us?

you can get it with following code:

```python
import platform
print(platform.machine())
```

Harvie, 2019-09-24:
Can you please try pip2 install again? @ptvoinfo added "aarch64" exception to setup.py.
Otherwise i will still need your platform.machine() output...

wabmester, 2019-10-24:
I'm having the same issue.   Here's my platform: armv7l

Harvie, 2019-10-24:
> I'm having the same issue. Here's my platform: armv7l

But setup.py currently contains conditional `"arm" not in platform_machine` which should prevent opencv from installing, because `armv7l` contains string `arm`. Are you sure your `platform.machine()` string is really armv7l? and that you are installing latest version?

Otherwise there is some problem with setup.py and we need to figure it out...

wabmester, 2019-10-24:
I don't know python or pip, but when I try to isolate that string from the setup script, it doesn't get evaluated as a conditional.    I.e., the string ends up literally as "opencv-python>=2.4 ; ("arm" not in platform_machine) and ("aarch64" not in platform_machine)"

I tried this in a couple different contexts, and got the same literal (using python 2.7.13)

Harvie, 2019-10-24:
Can you please try `pip install --upgrade pip` ?

wabmester, 2019-10-24:
"Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages "

(ran pip again just to be sure -- same error).   Just out of curiosity, can you point me to the doc for that setup feature.    I didn't see it when I looked for it.

Harvie, 2019-12-01:
> I don't know python or pip, but when I try to isolate that string from the setup script, it doesn't get evaluated as a conditional.

I think this is OK. the string should be evaluated by pip, not by setup.py.
Maybe the condition is not correctly formated, i don't know. But it seems to work most of the times... Any idea why your machine is different?

wabmester, 2019-12-01:
I'll dig into it some more, but apparently the conditional dependency feature is a "minefield:"
https://hynek.me/articles/conditional-python-dependencies/

Harvie, 2019-12-02:
@wabmester Can you please discuss this with setuptools guys?

https://github.com/pypa/setuptools/issues

Please show them our setup.py https://github.com/vlachoudis/bCNC/blob/master/setup.py
and describe how it still tries installing opencv on your machine even when disabled in setup.py

quinkennedy, 2020-02-16:
@wabmester did you create an Issue on the setuptools github?

I have run into the same problem (opencv tries to install despite being in the exclusion list for "arm"):
* Raspbian 9.11
* Python 2.7
* pip 20.0.2
* setuptools 44.0.0

I thought maybe it had to do with my setuptools version (was v32) but after upgrading, the installer still doesn't work. I have seen that Python 2.7 is now deprecated: https://www.python.org/doc/sunset-python-2/

quinkennedy, 2020-02-16:
In light of Py2 being deprecated, I decided to use py3

```
pip3 install --upgrade setuptools pip
pip3 install --upgrade bCNC
python3 -m bCNC
```

Perlovka, 2020-03-10:
So, python-opencv is not an optional dependency, as stated in README?

Tried to install and got
```
pkg_resources.DistributionNotFound: The 'opencv-python>=2.4' distribution was not found and is required by bCNC
```

quinkennedy, 2020-03-10:
@Perlovka it seems that the issue is that python-opencv is specified as an optional dependency, but setuptools for python 2.7 doesn't properly parse the dependecy list. AFAIK your options are:

* manually edit the configuration to completely remove the python-opencv dependency
* use python 3

Perlovka, 2020-03-10:
@quinkennedy, python 3 requires python-opencv as well, so removing python-opencv from setup.py is the only option.

Even more - there is no python-opencv package in my system (gentoo just build opencv with python modules if you need them), so this will fail even if opencv is installed.

bosd, 2020-11-13:
I think this one can be closed as well

closed 2020-11-13 by Harvie