vlachoudis/bCNC #1064: bCNC still doing anything

mirecta, 2018-11-24:
i cant do anything bCNC stil cyclic doing sometring run and stop after connect grbl machine no gcode file opened 
in bCNC console cyclic

```ok
Run ended
2018-11-24 17:26:30.516423
Current: 2 [2]  Completed: 100% [0s Tot: 0s ]
Grbl 1.1f ['$' for help]

ok
Run ended
2018-11-24 17:26:33.538935
Current: 2 [2]  Completed: 100% [0s Tot: 0s ]
Grbl 1.1f ['$' for help]
```

in linux console again and again 

```Controller state changed to: Idle (Running: True)
Ignoring machine stream request, because of state:  Idle True
Controller state changed to: Idle (Running: False)
Job done. Purging the controller. (Running: False)
Stream requested by CYCLE START machine button
Will clean after this operation
```

mirecta, 2018-11-24:
sorry my bug on my firmware grbl

closed 2018-11-24 by mirecta

Harvie, 2018-11-24:
V pohodě :-) Sounds like you had probe pin always active. I guess that chaning "invert probe pin" setting helpded.