vlachoudis/bCNC #1065: Random work and probe freezing

MrclaudsJHB, 2018-11-24:
Hi
Bcnc 0.9.14
Raspberry PI 3
Platform linux
Tk version: 8.6
Tcl version: 8.6
Issue:
When starting a program... The machine starts moving and then stops... Bcnc shows idle and nothing happens, I must stop and then press play/start again and then it proceeds normally...
It also does this randomly on probing and tool changes and is highly frustrating as when I press stop, I have to home everything again... 
Anyone have a similar issue? And a solution?

Harvie, 2018-11-24:
When was the last time you've updated bCNC? Sounds to me like #1061

MrclaudsJHB, 2018-11-24:
About a month ago?
Seems you have done some tweaks...
I'll load it up and see ;)

Harvie, 2018-11-25:
Have you tried new "pip" installation? https://github.com/vlachoudis/bCNC/wiki/OS