vlachoudis/bCNC #1068: Constant Crashing (I'm not sure how to title this).

ccomito1223, 2018-11-25:
GRBL0
grbl0 loaded
GRBL1
grbl1 loaded
SMOOTHIE
smoothie loaded
test super
test super
ctrlist
test super
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner self.run()
  File "C:\Python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "C:\bCNC-master\Sender.py", line 876, in serialIO self.serial.write(b"?")
  File "C:\Python27\lib\site-packages\pyserial-3.4-py2.7.egg\serial\serialwin32.
py", line 323, in write
    raise writeTimeoutError
SerialTimeoutException: Write timeout

Traceback (most recent call last):
  File "C:\bCNC-master\Utils.py", line 420, in __call__
    return self.func(*args)
  File "C:\bCNC-master\Sender.py", line 859, in stopRun
    self.purgeController()
  File "C:\bCNC-master\Sender.py", line 844, in purgeController
    self.control.purgeController()
AttributeError: ControlFrame instance has no attribute 'purgeController'

Harvie, 2018-11-25:
Duplicate of #1067

closed 2018-11-25 by Harvie

Harvie, 2018-11-25:
I think you don't have latest version of bCNC. This was already fixed in 0.9.14.42

```
# pip2 show bCNC
Name: bCNC
Version: 0.9.14.42
```