vlachoudis/bCNC #1070: Getting info when a gcode command finish execution

qvisionsa, 2018-11-28:
Hi, I'm a recently using your fabulous software and I'm trying to create a plugin.
In my plugin I'm sending commands like G0X0Y20 and I would like to ask you if there is a way to know when a command finish the execution before sending another.
I'm scanning a region to find points with my camera, but the code finish before the machine comes to a stop from the previous command. 
Is that possible?

Thanks 
Stelios Malakoudis

Harvie, 2018-11-28:
> I'm trying to create a plugin.

I am happy for your efforts! Please make sure that you are using latest git version of bCNC when writing plugin.

>  I would like to ask you if there is a way to know when a command finish the execution before sending another.

woah. this is tough question. yes and no. there is some kind of infrastructure for this, but i can't promise that it's not gonna change, because it's a mess and i don't like it. Like a week ago i was fixing some stuff related to this and i am not really happy with how it currently works...

> I'm scanning a region to find points with my camera

Are you using some kind of computer vision to automaticaly follow contours and digitize shape?

I am not really happy with how the automatic probing works now in bCNC and in a long run i want to modify the probing/scanning framework to allow more complex scanning schemes and usecases. Such as #946 #956 Maybe we can work together on creating some common infrastructure that would allow to easily implement "scanning plugins". However right now i am quite busy with making things calm and stable for 0.9.15 release (we really need this as last release was more than year ago...)

Can you please elaborate on what are you exactly trying to do and how you want to achieve that? I think we can discuss it to find the best solution to implement your plugin.

qvisionsa, 2018-11-28:
Thank you for your response! I'm really new to BCNC and Python(2.7) but I'm trying hard.
I would like to create a plugin to estimate the origin and rotation of a part (In my case a printed design) and then apply the transformations to g code file and cut it.
I'm using opencv with aruco markers to estimate the position and rotation. 
The first part of the plugin is finished. I can scan a small region to find the marker and reposition at 0,0.
This part is maybe working, because the distances are small.
To find the second marker that is about 500mm away, I'm getting out of code. I mean that my commands loop finished before reaching the marker. 
Is there a method or a Boolean variable that indicate when a gcode command finish?

Harvie, 2018-11-28:
In `Sender.py` i use following snippet `self.running == False and state in ("Idle"):`
But i hate how it works (basicaly you have to poll it in busy wait). So i have plans to change it to emulate synchronous behaviour.

> I would like to create a plugin to estimate the origin and rotation of a part (In my case a printed design) and then apply the transformations to g code file and cut it.

We already have very similar functionality. See probe->orient feature.

> I'm using opencv with aruco markers

Why do you want to use aruco markers? That sounds like too specific use case to me. Are you going to use dragknife to cut pre-printed sheets of paper? Why isn't the rotation consistent between individual prints?

qvisionsa, 2018-11-29:
I’m going to use drag knife and cut many identical printouts. Every printout that I put on the table, may have a slanted position. Looking on the UI of the program, I notice that the state change in Running, Idle etc. is there a state variable that holds this information?

qvisionsa, 2018-11-29:
Hi Harvie,
Today I will give a try to your suggestion and I will let you know.
I saw the dragknife feature, but I don't understand how it works. Is there any documentation for this?

Harvie, 2018-11-29:
> I saw the dragknife feature, but I don't understand how it works. Is there any documentation for this?

Not really yet, just few images: https://github.com/vlachoudis/bCNC/wiki/Drag-Knife

It compensates for the draknife tip offset. You enter how far is knife tip from rotation axis and apply that transformation on the desired shape.

qvisionsa, 2018-11-30:
Hi Harvie,
I was testing my code with your suggestion but I fail.
But I found in Sender module a function that returns the state of the controller. So I thought that this is a good point get a response when the controller "Run". 
I will give a try tomorrow and I will let you know. 
Do you implement the observable pattern in your code?

jschoch, 2019-01-28:
grbl doesn't really make checking on it's command status easy.

One approach may be to simply poll the current "dro" coordinates, return when they reach your intended position and then check for idle state.

Another more complex approach would be to create a state machine and then poll the states and state changes.