vlachoudis/bCNC #1071: How do I remove the separate Z-step size?

Lester444, 2018-11-29:
I find the separate Z-step height confusing to use with a pendant. I prefer to use the older version where all the axes use the same step size. Please see attached pic.

How do I go back to the previous display option?

[image]

Lester444, 2018-11-29:
How & where do I do that?

> Just set the Z step to "XY"

Harvie, 2018-11-29:
[image]

If you set it like this, it will take Z step from XY step, so you will only need to set one value.

Lester444, 2018-11-29:
I checked just now, that "XY" option is not on the drop-down menu options. All I see are numbers, no "XY".

 I am using bCNC 0.9.11, downloaded from the released versions.

Harvie, 2018-11-29:
I see. XY option is in git and will be available in upcoming 0.9.15 release.
If you use 0.9.11 then just open your bCNC.ini and replace `stepz...` with `;stepz...` (or completely delete the line beginning with stepz)

Lester444, 2018-11-29:
I have tried both:
1. replaced zstep = 1  with  ;zstep = 1
2.  completely deleted this line

It does not work, I still see the separate z-step size. Still no "XY" option.

This part of bCNC.ini
[Control]
step   = 1
wcs    = 54
step1  = 0.1
step2  = 1
step3  = 10
swap   = 0
steplist = 0.001 0.005 0.01 0.05 0.1 0.5 1 5 10 50 100 500
zsteplist = 0.001 0.005 0.01 0.05 0.1 0.5 1 5 10 25

Lester444, 2018-11-29:
changing
zsteplist = 0.001 0.005 0.01 0.05 0.1 0.5 1 5 10 25
to
;zsteplist = 0.001 0.005 0.01 0.05 0.1 0.5 1 5 10 25

......worked fine. Thank you.