vlachoudis/bCNC #1072: Connect to remote motion controllers over IP network

Harvie, 2018-12-01:
There is interresting wifi and bluetooth enabled port of GRBL to ESP32:
https://github.com/bdring/Grbl_Esp32

So it came to my mind, that we can support this. But then i found, that pyserial library already knows how to connect to remote serial using this URL syntax `rfc2217://localhost:2217`

[image]

I want to test the compatibility with various connection options, like https://github.com/bdring/Grbl_Esp32/issues/60

mstrens, 2019-03-02:
First thanks for all your work developping BCNC.

I need some help. 
I am currently writing a GRBL controller that should run on a ESP32.
Here the link of the project: https://github.com/mstrens/grbl_controller_esp32

The idea is to connect the ESP32 to a touch screen and a SD card support in order to control another micro computer running GRBL. So GRBL could be controlled without the need of a pc connection.
Still in order to configure GRBL and if the user want to use the keyboard of the pc to send some GRBL commands, I would also like to allow the user to use BCNC with a telnet protocol (over Wifi). 

On ESP32 site, I wrote the telnet server and it seems to work. If I open a telnet session on my windows pc, I can exchange telnet messages between PC and GRBL.

Now, I would like to test it with BCNC.
I am running Windows 10 (64bit).
I just dowloaded the windows exe version of BCNC from this link https://github.com/vlachoudis/bCNC/releases
I selected the latest release (0.9.14.52).
Installation was successful and I can run BCNC.
Still I can't open a telnet session.
In the window with the serial parameter, in "Port" I entered "rfc2217://192.168.1.5:23
192.168.1.5 is the ip adress of my ESP32 and 23 is the standard application port for telnet used in the ESP32.
When I try to open the connection I immediately get an error in a small window.
Titel of window is "Error opening serial"
Message is "Invalid URL, protocol 'rfc2217' not known"

Could it be that I have to install some additional package (pyserial???).
Is this package not included in the "exe" version that I downloaded?
If I have to install it, please explain me how and where.

Harvie, 2019-03-02:
> I am currently writing a GRBL controller that should run on a ESP32.

Why not join efforts with https://github.com/bdring/Grbl_Esp32 ?
They have the same goal, but without using separate uC for GRBL, they run both on ESP32...

> Message is "Invalid URL, protocol 'rfc2217' not known"

Try upgrading to latest pyserial, eg.: `pip install --upgrade pyserial`
Unfortunately i had to put very old pyserial version to bCNC install script, because more recent versions are not working on Windows XP. I have to figure out how to enforce this version on Windows XP only. But now you should be able to upgrade pyserial once bCNC is installed...

mstrens, 2019-03-02:
Thanks for your fast answer.
I saw the project about running GRBL on a single ESP32.
Still I am not sure that a single ESP32 can handle all tasks (TFT, touchscreen, SD card, wifi,...)  and keep step pulse at the rigth time.
Furthermore the ESP32 does not have many pins to support all functions in GRBL + the screen/touchscreen/Sd card.
I wanted a solution that can run in standalone too (no connection with a pc).
Therefore I took the option to use 2 computers:
- a stm32 bluepill to run GRBL
- a ESP32 to manage the screen/touch screen/sd card and (optional) connection with pc based on USB or Wifi. I use the ESP as webserver to allow to upload/delete/list files on the SD card and as telnet server to control GRBL.
The project is open source and is available at the link here above.
Feel free to use part of it if you want.

I just deleted my BNC installation (based on the exe file) and I followed the other solution.
So I installed python 2.7 and bnc
It installed 
- bcnc 0.9.14.53 
- pyserial 3.0.1
- numpy 1.16.2
- pillow 5.4.1

Now, when I try to connect, I get another error:
"Remote does not seem to support RFC2217 or BINARY mode [we-BINARY:False(INACTIVE), we-RFC2217:False(REQUESTED)]"

Perhaps I have to upgrade.
I will try to morrow.

mstrens, 2019-03-02:
I already tried to upgrade pyserial with the command your provided.
Upgrade is done but I still get the same error.
Note: in ESP32, I get a message saying that it is connected to telnet but it is not the case in bCNC

Harvie, 2019-03-02:
Are you sure that your device implemented  RFC2217? I think you should rather try `socket://`

mstrens, 2019-03-03:
I tried socket:// and I now get the connection.
I get e.g. the GRBL status Idle.
Still it is not stable. After a few second ESP32 locks and BCNC says it is not connected.

I will further look on ESP32 side for a bug.

mstrens, 2019-03-03:
It seems OK.
In ESP32 I have a menu to select the source of GRBL commands (Standalone/USB/telnet).
When ESP32 is connected to bCNC but is in standalone, I had to clear continously the telnet input buffer in ESP32 (and discards the char); otherwise some buffer becomes full and ESP32 locks.
Thanks for your help.

Harvie, 2019-03-03:
```
https://github.com/bdring/Grbl_Esp32
https://github.com/mstrens/grbl_controller_esp32
```

Since there are already two projects doing this, i think we should agree on some basic authentication mechanism:

I think that bCNC can send something like this immediately after connection/reset:

```
(USER cnc)
(PASS cncpwd)
```

It will be ignored by other GRBL implementations, since it's a comment, but networked setups can use it to prevent random people from messing with your CNC.

misan, 2019-03-03:
@Harvie Is that sent over a plain TCP connection? If so, it seems weak (but sure better than nothing).

mstrens, 2019-03-03:
In my project I do not think that it is really needed.
Currently it works lke this:
ESP32 checks (in main arduino loop) if a telnet connection exist and display a status on TFT.
ESP32 accepts only one connection (reject a new one if already connected).
When ESP32 is in mode "standalone" or "USB", it just read but discards all chars coming from telnet.
When the user want to use the telnet, it has:
- first to open the telnet transaction on bCNC side. Status of telnet on TFT says if telnet is OK.
- second it has to select (on ESP32 menu) the option to enter "telnet" mode.
From this time, ESP32 just forwards all chars between GRBL and bCNC.
To stop the "telnet" mode, user has to select a "Stop PC" option in ESP32 and this enter the "Standalone" mode again.
If the user selects the option "telnet" mode while there is no connection,  then ESP32 gives an alert and stay in "standalone" mode.
I presume there is no big risk that random people can mess with CNC because there is only one telnet connection at a time and the user has to select an option in a menu to let it have the control of the CNC .

mstrens, 2019-03-03:
Just for info:
I accept to open the telnet connection even in standalone mode because it allows me to forward all messages coming from GRBL to a telnet session (as said, incomming telnet messages are discarded in standalone mode). 
Currently, I do not store/display the GRBL message on the ESP32 TFT
The TFT gives only the GRBL status, the WPOS, MPOS and the last error or Alarm.
So I though, it could be useful for debuging while in standalone.

Harvie, 2019-03-03:
> Is that sent over a plain TCP connection? If so, it seems weak (but sure better than nothing).

It is weak, but they might add TLS support to serial in future. It's more for preventing random accidents rather than perfectly securing your CNC. But you should be OK if you are in the same LAN as your CNC. Also people are working on QUIC protocol, which is probably going to replace TCP directly and it will be encrypted by default, so in future there might not be need for TLS at all.

dylanpil, 2021-12-03:
@Harvie I have been enjoying using Bcnc for my cnc, but I have continuously been changing the Microcontroller.
I started with arduino Uno, then moved onto arduino mega with onboard esp. And now have moved onto only the esp32 using Barts initial project and now his newer project FluidNC.
https://github.com/bdring/Grbl_Esp32
https://github.com/bdring/FluidNC
I had great success using the arduino mega and **wifi** using GRBL of Mega and ESP-Link
https://github.com/jeelabs/esp-link
Unfortunately since I've upgraded to the ESP32 and using Barts projects, I cant seem to get a good result. using GRBL_Esp32 seems to work fine but the drops part of the gcode while running.
When using FluidNC, there seem to be major delays in response time but it seems like it keeps the data, but i haven't tested this enough. 
I have tried connecting to them over my wifi network, and directly connected to them as an access point. and the issues seem to be the same. I still have to try over a data cable and see if there are any delays via the cable.
Any ideas would be appreciated.

misan, 2021-12-03:
> drops part of the gcode while running
What does that mean? Some of the GCODE is ignored and then it continues business as usual?

dylanpil, 2021-12-03:
it cuts the shape but seems to miss steps in odd places in odd intervals so the parts dont always complete at starting path of shape

misan, 2021-12-03:
@dylanpil that problem seems more an error condition than a communications problem. I would check the open issues with grbl_ESP32 and, AFAIK, it could be an issue with the ESP32 power supply, as it has spikes of consumption that a not-so-good power supply might not enjoy, triggering some random behavior in these moments.

dylanpil, 2021-12-04:
@misan thanks, i would love to sort this out so i'm also trying with the grbl_ESP32 team.
My power supply is a decent 32v, which the CNC has built in, it connects to the 6 pack cnc controller which can take 12-36v.
https://github.com/bdring/6-Pack_CNC_Controller
my motor drivers are also connected to this power supply.
I could try running it off a separate decent power supply.

misan, 2021-12-04:
@dylanpil I meant the ESP32 supply voltage perhaps not being stable enough, not the motor's supply. If possible try to power it off a power bank to see if that helps.

dylanpil, 2021-12-05:
@misan I understood what you were saying, and as soon as I'm able to test a separate power bank, I'll give you feedback. thanks for your guidance.