vlachoudis/bCNC #1073: Wrong coordinates returned from CNC.vars

qvisionsa, 2018-12-02:
I'm writing a plugin and in some point Im trying to get working and machine coordinates, but the results are different from the real position. I'm getting the values from CNC.vars wx, wy, mx and my. Am I wrong?
Is there any other method to get these values accurately?

MARIOBASZ, 2018-12-02:
I have worked well with wx wy, what is the problem? Good for contributing a plugin! Curiosity, what is it about?

qvisionsa, 2018-12-02:
I'm working on markers scanning and code rotation. I'm not finished yet. I'm very new in python.
The fact is that after finishing a move (like G1 X0 Y460) or after sending a command to set x=0 and y=0, I set some variables to get working position or machine position. The values returned are from CNC.vars["wx"] etc. are wrong !

Harvie, 2018-12-02:
> The values returned are from CNC.vars["wx"] etc. are wrong !

I think this might be a feature rather than bug.
Do you realize that by reading CNC.vars you are reading variable in bCNC, not status of GRBL. bCNC periodicaly sends "?" command to which GRBL responds with so called "status report". Once bCNC receives this status report, it updates CNC.vars. If you read CNC.vars immediately after sending the commands, it does not have enough time for this to happen.

I am not really sure about this. But bCNC might have some method which will cause to wait until status report is received. Also note that you have to wait for GRBL to become idle before waiting for status report.

This document might explain the inner workings bit better:
https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl

closed 2018-12-02 by Harvie