vlachoudis/bCNC #1074: Interface is unreadable on a 4K display

vatral, 2018-12-03:
Everything is absolutely tiny.

The program should detect and automatically compensate for high DPI mode.

Harvie, 2018-12-04:
Have you tried enabling double sized icons in bCNC.ini?

```
[CNC]
doublesizeicon = 1
```

vatral, 2018-12-07:
Okay, that and changing the fonts configuration improved matters quite a bit, thanks. 

However, the default should be automatically readable. It took me a good while to figure out there's a difference between a font set to 12 and one set to -12, because the "-" is that tiny on the screen. 

I do appreciate that it's customizable enough to make it work, though :)

Harvie, 2018-12-07:
Yes, i think it should be possible to autodetect HiDPI displays. Maybe based on window size.
But i don't have time to do this. Pull requests welcome.

leofaoro, 2018-12-21:
sure will do