vlachoudis/bCNC #1076: Dangerous behavior of Autolevel with G53

MARIOBASZ, 2018-12-04:
Autolevel is incompatible with G53.
Finish the sketch with
G53 G0 Z-4
the machine went directly to Z-4 in the G54 work case. Small difference of 100mm

Lester444, 2019-01-08:
I have also experienced this dangerous behavior, but at the start of my program. It crashed my tool into the workpiece  (something like G54 G0 Z-1) instead of going to G53 G0 Z-1 which is what is in the g-code file. It only happens when the probe file (filename.probe) is active. If I disable the probe file then machine behavior is normal.

DanielMiner, 2021-09-07:
This is also a problem for me.  It crashed my tool into the work piece - fortunately a scrap / test piece. 

Is there any hope of this dangerous bug getting fixed?