vlachoudis/bCNC #1081: Solution to re-execute cut

MARIOBASZ, 2018-12-05:
I do not know why the conflict is generated with (stop considering it a comment?)
newblock.append ("(pass% f)"

It is solved if it is corrected
  newblock.append ("((pass% f))"

Harvie, 2018-12-06:
I've just tried and re-cutting works for me. Have you tried vlachoudis master? Can you send me g-code that causes problems and how to do it?

MARIOBASZ, 2018-12-06:
m missing add operation with islands.
I chose islands, (or tabs).
Here is an example.
With individual block also happens
I cut z-1
Ok, I must correct depth
I cut to different Z and .....
Houston are in trouble ...... we have lost some teammates.

[Problem_Cut.zip](https://github.com/vlachoudis/bCNC/files/2653846/Problem_Cut.zip)

MARIOBASZ, 2018-12-07:
Now that I remember, the problem existed with tabs before the implementation of islands. One solution is not to overwrite the original, but to leave the block disabled, to rebuild a new cut with islands.

Harvie, 2018-12-07:
Yes. I tested and it happens only with tabs.

Note that i've added "flatpath" [image] plugin in order to help debug re-cut problems. If you select cutted path and then use flatpath on it. It should give you original 2D path. In this case it is broken, so we know that something went wrong.

MARIOBASZ, 2018-12-07:
ok I did not realize using flat path