vlachoudis/bCNC #1091: Milling with smoothieware results in command submission stopping after a while

vatral, 2018-12-09:
Scenario:

Proxxon MF70 with a Smoothieboard running Smoothieware, using firmware-cnc-latest (currently latest available, uploaded 9 days ago)

I am running the latest code installed with pip from git on Fedora (0.9.14, 5 Feb 2018)

1. I connect (serial port, SMOOTHIE)
2. Home
3. Load a gcode file generated in Fusion 360
4. Run

Result:

A few moves get executed, until everything just stops. The endstops aren't being hit, there's no alarms or errors displayed. 

The status says: Current 183 [3656] 
Controller buffer fill is at 93%

The status is "Idle"
The "Start" button remains greyed out.

On the terminal where I ran the program it says:

```
new-config Utils <ConfigParser.ConfigParser instance at 0x7f7aad9435f0>
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/threading.py", line 804, in __bootstrap_inner
    self.run()
  File "/usr/lib64/python2.7/threading.py", line 757, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/site-packages/bCNC/Sender.py", line 738, in serialIO
    elif self.mcontrol.parseLine(line, cline, sline):
  File "/usr/lib/python2.7/site-packages/bCNC/controllers/_GenericController.py", line 184, in parseLine
    self.parseBracketAngle(line, cline)
  File "/usr/lib/python2.7/site-packages/bCNC/controllers/SMOOTHIE.py", line 53, in parseBracketAngle
    d= { a: [float(y) for y in b.split(',')] for a, b in [x.split(':') for x in l[1:]] }
  File "/usr/lib/python2.7/site-packages/bCNC/controllers/SMOOTHIE.py", line 53, in <dictcomp>
    d= { a: [float(y) for y in b.split(',')] for a, b in [x.split(':') for x in l[1:]] }
ValueError: invalid literal for float(): 40.010.

Will clean after this operation

```

The gcode file doesn't contain any '40.010' anywhere.

Further command submission fails. For instance I can't send an M999 to Smoothie.

So, suspecting that there's something wrong in the smoothie code, I tried connecting in GRBL1 mode. That actually worked better. The machine completed the ellipse this time, and went a tiny bit further:

Here's the end of the gcode window:

```
G18G2X32.051Z-0.7K0.3
ok
G1Z15F1000
ok
G17
ok
M5
ok
G28G91Z0
ok
G90
ok
M0
ok
T1
ok
S20000M3
ok
M106
ok
G0X5.225Y5.533F1000
ok
G0Z15
ok
G1Z5
ok
G1Z2.6F50
ok
```

Here's the buffer:

```
G1X5.25Y5.546Z2.53F300
G1X5.307Y5.583Z2.5
G3X4.729Y6.337Z2.448I-0.289J0.377
G3X5.307Y5.583Z2.396I0.289J-0.377
```

The terminal contains:

```
python2 -m bCNC
new-config Utils <ConfigParser.ConfigParser instance at 0x7fb7252865f0>
Controller state changed to: Alarm (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Home (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Will clean after this operation
Controller state changed to: Idle (Running: True)
Controller state changed to: Run (Running: True)
Controller state changed to: Idle (Running: True)
Controller state changed to: Hold (Running: True)

```
A bunch of those is probably me jogging things around. The hold is also mine, but was issued only after the machine had stopped moving for quite some time.

I had inserted a manual tool change with Fusion 360, and wasn't sure what was going to happen at that point (I'm still very new to this), so there may be going something wrong with that. The status simply goes to "Idle", there's no message from the program of any sort.

Also, earlier I tried using Pronterface to submit gcode instead. That seemed to work perfectly fine, but it doesn't seem like the best fitting tool for the task.

Here's the gcode I'm sending:

[Vika.zip](https://github.com/vlachoudis/bCNC/files/2661018/Vika.zip)

So it seems there may be several issues here:

1. Smoothie support has some sort of bug in it
2. There are conditions in which bCNC can stop working and not say anything to the user
3. There may be an issue with different implementations of a manual tool change. It looks like Fusion 360's approach is an M0 rather than an M6.

Thanks!

Harvie, 2018-12-09:
Was this working in some older version? Can you please try eg. 0.9.11 release, so i can track if there were any changes?

Problem is that i don't have any board capable to run smoothie in order to debug this. I would be super happy if somebody can debug this for me? all relevant code should be now in `bCNC/controllers/SMOOTHIE.py` and  `bCNC/controllers/_GenericController.py` files.

I guess there's some problem with parsing, where float() gets "40.010." instead of "40.010". dunno.

If there's someone with smoothie and python knowledge. Can you please take a look at following part of SMOOTHIE.py ?

```python
        def parseBracketAngle(self, line, cline):
                # <Idle|MPos:68.9980,-49.9240,40.0000,12.3456|WPos:68.9980,-49.9240,40.0000|F:12345.12|S:1.2>
                ln= line[1:-1] # strip off < .. >

                # split fields
                l= ln.split('|')

                # strip off status
                CNC.vars["state"]= l[0]

                # strip of rest into a dict of name: [values,...,]
                d= { a: [float(y) for y in b.split(',')] for a, b in [x.split(':') for x in l[1:]] }
                CNC.vars["mx"] = float(d['MPos'][0])
                CNC.vars["my"] = float(d['MPos'][1])
                CNC.vars["mz"] = float(d['MPos'][2])
                CNC.vars["wx"] = float(d['WPos'][0])
                CNC.vars["wy"] = float(d['WPos'][1])
                CNC.vars["wz"] = float(d['WPos'][2])
                CNC.vars["wcox"] = CNC.vars["mx"] - CNC.vars["wx"]
                CNC.vars["wcoy"] = CNC.vars["my"] - CNC.vars["wy"]
                CNC.vars["wcoz"] = CNC.vars["mz"] - CNC.vars["wz"]
                if 'F' in d:
                        CNC.vars["curfeed"] = float(d['F'][0])
                self.master._posUpdate = True
```

vatral, 2018-12-09:
I've just started messing with this stuff, so I've never tried any earlier versions.

I'll give 0.9.11 a try, and see if I can poke around in the source a bit (I'm not particularly skilled in Python though)

By the way, I tried interceptty, and bCNC refuses to connect to the port (illegal operation I think -- unfortunately right now I can't give it a try to check, but I'll make a proper report of that later)

Harvie, 2018-12-09:
> bCNC refuses to connect to the port 

probably problem with permissions of that tty in /dev

I've found that you can debug serial like that without interceptty. put this to port input in bCNC:

`spy:///dev/ttyUSB1?raw`

Harvie, 2018-12-09:
> (I'm not particularly skilled in Python though)

I was never a python guy. But then i started tweaking bCNC, so now i can add python to my CV lol :-D
But unfortunately i only use bCNC with GRBL...

vatral, 2018-12-11:
Okay, I think I'm starting to get an idea. I'm running from the source tree checked out from git now, 78aa077aac0050bc6935a9c54f3ba04d933aef26

No parse error so far, but it did still stop. Here:

```
?<Run|MPos:32.0509,0.3000,6.3883|WPos:32.0509,0.3000,14.3883|F:300.0,1000.0,100.0>
ok^M
ok^M
ok
G3X4.729Y6.337Z2.448I-0.289J0.377
ok
ok
ok
G3X5.307Y5.583Z2.396I0.289J-0.377
ok
ok^M
ok^M
G3X4.729Y6.337Z2.344I-0.289J0.377
?ok^M
<Run|MPos:32.0509,0.3000,7.0000|WPos:32.0509,0.3000,15.0000|F:0.0,300.0,100.0>
?<Run|MPos:31.7700,0.3547,7.0000|WPos:31.7700,0.3547,15.0000|F:356.6,300.0,100.0>?<Run|MPos:32.0509,0.3000,6.3883|WPos:32.0509,0.3000,14.3883|F:300.0,1000.0,100.0>
ok^M
ok^M
ok
G3X4.729Y6.337Z2.448I-0.289J0.377
ok
ok
ok
G3X5.307Y5.583Z2.396I0.289J-0.377
ok
ok^M
ok^M
G3X4.729Y6.337Z2.344I-0.289J0.377
?ok^M
<Run|MPos:32.0509,0.3000,7.0000|WPos:32.0509,0.3000,15.0000|F:0.0,300.0,100.0>
?<Run|MPos:31.7700,0.3547,7.0000|WPos:31.7700,0.3547,15.0000|F:356.6,300.0,100.0>
... 
?<Run|MPos:5.4434,6.1841,-5.6264|WPos:5.4434,6.1841,2.3736|F:300.0,300.0,100.0>
?<Run|MPos:4.8673,6.4003,-5.6506|WPos:4.8673,6.4003,2.3494|F:300.0,300.0,100.0>
?<Idle|MPos:4.7290,6.3370,-5.6560|WPos:4.7290,6.3370,2.3440|F:300.0,100.0>
?<Idle|MPos:4.7290,6.3370,-5.6560|WPos:4.7290,6.3370,2.3440|F:300.0,100.0>
```

Once the status starts returning Idle, things get stuck. Here's the full spy log of that session, including the initial connection, homing, jogging, and the entire job until failure:

[bCNC.log](https://github.com/vlachoudis/bCNC/files/2669681/bCNC.log)

I also did a full successful run, once (with a slightly different file than above). Here's what happened at the end. This is the very bottom of the commands window:

```
G1Z15F600
ok
M107
ok
G28G91Z0
ok
G90
ok
```

This is exactly what is right at the end of the file, so good. However, bCNC still doesn't think the job is over. The buffer window contains:

```
G28G91X0Y0
G90
```

So here's my take: With Smoothie, there are at least some cases in which either Smoothie doesn't confirm a command, or bCNC misses the confirmation. Then at some random point in the middle of the job, bCNC is waiting for more confirmations, while Smoothie is waiting for more work, and nothing goes anywhere.

Pending things to figure out:

1. Does this happen consistently? When it breaks, is it always at the same point? Is it a timing/threading problem, or a protocol problem?
2. Is there a confirmation issue with some particular command?
3. Is it a Smoothie problem, or bCNC problem?
4. Could there be a problem with the data on the wire? This might explain the random parse error that now isn't happening. Or that could be something else.

Harvie, 2018-12-16:
Is this related to #1100 ?

Harvie, 2018-12-16:
Is there still some practical issue with log not being up to your expectations?

I know the sender code is a big mess right now. I've started to untangle the code a bit... But it will take some time... In other words: if you have time to do major sender cleanup, i will happily pull your changes. I don't have time for this. So unless this causes problems, i am afraid there's nothing i can do.

vatral, 2018-12-16:
So my original problem is that I generate a .nc file, open it, run it, and milling proceeds to a point, until sometimes it just stops. bCNC keeps running, the interface remains responsive, but bCNC doesn't submit any more gcode to the machine and shows an "Idle" status in the middle of an active job.

At that point all I can do is to abort the job entirely. There's no way to resume it.

Now the $G issue definitely accounted for a large part, if not all of that. The thing is that I wasn't sure why even accounting for the $G bug, it was stopping precisely at that point.

So I just poked around the source a bit more and just realized: the buffer limit is in bytes. If a confirmation frees up the space taken by a short command, and then a longer one must take place, then indeed it's possible to have a situation where an "ok" still doesn't allow to proceed.

So after looking at that, things make a lot more sense to me and it's quite likely everything is fixed at this point. I would say go ahead with that release

Harvie, 2018-12-16:
Ok. If you feel like there is some special case in which it stops responding, you can try to do torture test of bCNC to find out how to trigger that. Please let us know if you find something.

vatral, 2018-12-16:
I already did some preliminary testing, and the results look good so far.

I will definitely be doing more testing soon.

Harvie, 2018-12-17:
Thanks for your efforts.