vlachoudis/bCNC #1093: Autolevel for MPos

wuerfelfreak, 2018-12-10:
Hi there,
I have a machine with a not quite leveled wasteboard. I didn't (and do not plan to) level my wasteboard.
Instead I wanted to use bCNC's autolevel function to compensate for this offset.
Then if i bolted down my stock, I could face it paralel.

But Autolevel saves the data in WPos, so that I would have to change that every time I change the position of my stock. I would need it as MPos so it stays relative to the maschine frame.
Is there a way to change that or is this a missunderstanding?

Any tips are appreciated.
Thank you very much.

MARIOBASZ, 2018-12-10:
Are you sure you save it in wps? in my head I have the impression that is stored in the coordinates of naquina

wuerfelfreak, 2018-12-11:
I looked at it again today:
Every time i change the WPos X/Y it moves the autoleveling points shown on the screen.
If you scanned the surface of a curved stock, this makes sense, but not when you scan your wasteboard.
Can I change the positioning?

Harvie, 2018-12-13:
Autoleveling is always done for workpiece. If you want to do persistent autoleveling of your machine, i guess we might add way to translate autolevel date do different location.

As workaround. Right now, you can set WPOS zero to zero of your machine. And use "move" to move the toolpaths to absolute position.