vlachoudis/bCNC #1094: Connecting to interceptty gives "[errno 22] Invalid argument"

vatral, 2018-12-10:
I've been trying to do some debugging of issues with the smoothieboard with the suggested method of using interceptty. Unfortunately it doesn't connect.

Immediately after trying to connect, I get: "[Errno 22] Invalid Argument"

It looks like the issue is with:

`ioctl(5, TIOCMBIS, [TIOCM_DTR])   = -1 EINVAL (Invalid argument)
`

I'm not sure if this is a bug in interceptty due to not supporting a required ioctl, or in bCNC, due to not ignoring the lack of something the suggested interceptty doesn't provide.

```
$ groups
vadim wheel dialout mock libvirt

$ sudo interceptty -l /dev/ttyACM_cnc  /dev/ttyACM_cnc_intercept | interceptty-nicedump 2>&1 | tee dump.log                                                                                                                                         

$ ls -l /dev/ttyACM_cnc_intercept  
lrwxrwxrwx. 1 root root 10 Dec 10 20:15 /dev/ttyACM_cnc_intercept -> /dev/pts/4

$ ls -l /dev/pts/4
crw-rw----. 1 root dialout 136, 4 Dec 10 20:15 /dev/pts/4

$ ls -l /dev/ttyACM_cnc
lrwxrwxrwx. 1 root root 7 Dec 10 19:56 /dev/ttyACM_cnc -> ttyACM1
 
$ ls -l /dev/ttyACM1   
crw-rw----. 1 root dialout 166, 1 Dec 10 20:06 /dev/ttyACM1

```

strace:
```

42207 10285 openat(AT_FDCWD, "/dev/ttyACM_cnc_intercept", O_RDWR|O_NOCTTY|O_NONBLOCK <unfinished ...>
42208 10300 <... socket resumed> )            = 4
42209 10300 setsockopt(4, SOL_SOCKET, SO_REUSEADDR, [1], 4) = 0
42210 10300 bind(4, {sa_family=AF_INET, sin_port=htons(8080), sin_addr=inet_addr("0.0.0.0")}, 16 <unfinished ...>
42211 10285 <... openat resumed> )            = 5                                                                                                                                                                                                                                 
42212 10285 ioctl(5, TCGETS <unfinished ...>
42213 10300 <... bind resumed> )              = 0
42214 10285 <... ioctl resumed> , {B38400 -opost -isig -icanon -echo ...}) = 0
42215 10300 futex(0x5624e8aaee10, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, 0xffffffff <unfinished ...>
42216 10285 futex(0x5624e8aaee10, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
42217 10300 <... futex resumed> )             = -1 EAGAIN (Resource temporarily unavailable)
42218 10285 <... futex resumed> )             = 0
42219 10300 getsockname(4, {sa_family=AF_INET, sin_port=htons(8080), sin_addr=inet_addr("0.0.0.0")}, [16]) = 0
42220 10285 ioctl(5, TCGETS <unfinished ...>
42221 10300 futex(0x5624e8aaee10, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, 0xffffffff <unfinished ...>
42222 10285 <... ioctl resumed> , {B38400 -opost -isig -icanon -echo ...}) = 0
42223 10285 ioctl(5, SNDCTL_TMR_START or TCSETS, {B115200 -opost -isig -icanon -echo ...}) = 0
42224 10285 ioctl(5, TCGETS, {B115200 -opost -isig -icanon -echo ...}) = 0
42225 10285 futex(0x5624e8aaee10, FUTEX_WAKE_PRIVATE, 1) = 1
42226 10300 <... futex resumed> )             = 0
42227 10285 ioctl(5, TIOCMBIS, [TIOCM_DTR])   = -1 EINVAL (Invalid argument)

```

I have a workaround with the spy:/// port URL suggested in another bug, so this isn't a critical issue. However, it is a problem with the debugging method suggested right at the bottom of the github page:

https://github.com/vlachoudis/bCNC

Harvie, 2018-12-13:
Yes, i think this method is obsolete and i will remove it from documentation. spy:// is way to go now, as interceptty works only on linux, while spy works on all platforms.

however it should work. try to use something more conservative for naming. like /dev/ttyUSB9 instead of /dev/ttyACM_cnc. also check your permissions.

vatral, 2018-12-13:
The permissions are quoted above, they are both:

`crw-rw----. 1 root dialout`

and I am in the dialout group.

The naming works just fine for me without interceptty. bCNC finds and connects to /dev/ttyACM_cnc just fine. 

It also should be supported, as fixed names via udev are a standard Linux feature that I very much want in my case. That computer also connects to 3D printers and Arduino dev boards and it's a pain to deal with the randomly changing port names.

Harvie, 2018-12-13:
> That computer also connects to 3D printers and Arduino dev boards and it's a pain to deal with the randomly changing port names.

USB is freaking hell. I have 16 usb-to-serial converters in single server. As long as they are from single manufacturer, there is no chance telling them apart. You can try to tell them by which USB socket they connect to. But it's not really practical when you have staff changing the connections all the time. Plus they are connected in multiple USB hubs, which can't be told one from another for very same reason. Hell hell hell.

vatral, 2018-12-16:
Closing, since now it seems obsolete

closed 2018-12-16 by vatral