vlachoudis/bCNC #1096: Where is the file MyClamps.dxf for the tutorial?

jrlandau, 2018-12-13:
I would like to execute the MyClamps tutorial, but can't find the MyClamps.dxf to be loaded.

Harvie, 2018-12-17:
I don't know... The tutorial is rather old. You might ask @vlachoudis, but he's rather busy.
The tutorial states that it's very easy to create that file in librecad and i have to agree. Just rectangle, circle and slot. I can create the file for you if necessary, but i highly reccomend you to create it yourself. If you don't know how to use librecad, you can learn it in minutes and you will have powerfull tool for your further CNC work.

Harvie, 2018-12-17:
BTW i think we might add some "example library" to bCNC to automaticaly generate various simple shapes (such as clamps, stars, hearts, etc...) for given dimensions