vlachoudis/bCNC #1100: $G command is treated as unconfirmed

vatral, 2018-12-15:
I'm working on figuring out my current issues with Smoothieboard.  One of them seems to be that bCNC deems some commands not to have completed, keeps them in the buffer, and work ends up stalling as they pile up.

Here's one I found that works predictably: `$G`

Reproduction:

1. Start the program
2. Home

From the protocol log:

```
?$H
<Idle|MPos:0.0000,0.0000,0.0000|WPos:0.0000,0.0000,0.0000|F:4000.0,100.0>
$G
?<Home|MPos:0.0000,0.0000,0.8120|WPos:0.0000,0.0000,0.8120|F:240.0,4000.0,100.0>
...
?<Home|MPos:0.0000,0.0000,39.3253|WPos:0.0000,0.0000,39.3253|F:120.0,120.0,100.0>
ok
?[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F4000.0000 S0.8000]
<Idle|MPos:0.0000,0.0000,39.0000|WPos:0.0000,0.0000,39.0000|F:4000.0,100.0>
```

Buffer:
`$G`

Issuing a $G by hand has the same effect, the $Gs keep piling up in the buffer.

The GRBL docs (https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands) suggest that this is correct:

> 
>  When sending this command to Grbl, it will reply with a message starting with an [GC: indicator like:
> 
> [GC:G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 S0.0 F500.0]

However, I do not have a Grbl board currently. Could somebody confirm whether Grbl says "ok" afterwards, and whether there is a standard that specifies the correct answer? I'm not sure whether to fix smoothie or bCNC, but I guess currently a smoothie-specific workaround in bCNC might be in order.

vatral, 2018-12-15:
Further testing suggests that this may be the root of my issues:

I tried a test file made of many repetitions of this:

```
G0 X1.0 Y1.0
G0 X2.0
G0 Y2.0
G0 X1.0
G0 Y1.0

```

This works perfectly fine. 

But do a single $G before that, and things get out of whack. The first $G is deemed unconfirmed, then the "ok" to a G0 gets treated as a confirmation to the $G, and at the end there are G0 commands in the buffer and bCNC is stuck for them to be confirmed.

So it appears that at least $G and maybe other commands have the confirmation mechanism broken. This may be Smoothie's or bCNC's fault, I'm not sure which.

Harvie, 2018-12-16:
> before closing it may I suggest to add this one to the FAQ or somewhere else in the documentation in case someone else runs into it?

which releases of smoothie are affected by this?

vatral, 2018-12-16:
Smoothieware unfortunately doesn't seem to have numbered releases, or version numbers at all.

The affected releases would be anything released after Sep 30 2018. 

The first fixed version was uploaded in commit commit a555b096, on Sat Dec 15 22:20:56 2018 +0000

I've tested it, and the $G issue is fixed.

Harvie, 2018-12-16:
> Smoothieware unfortunately doesn't seem to have numbered releases, or version numbers at all.

I am not sure what to add to the FAQ. It's kinda hard if they don't have versioning scheme. I guess i will just add notice, that users should always use LATEST version of controller firmware, especialy smoothieware.

I will close this for now. As there is no real problem in bCNC.

closed 2018-12-16 by Harvie