vlachoudis/bCNC #1102: Delay in tool change spindle start

leofaoro, 2018-12-17:
Hello
I use the tool change feature, and when I am done with putting the new tool in, the spindle starts and quickly moves to the workpiece without having spun to the target spindle speed. Would you please add a delay command of 5s or so so the spindle can speed up before plunging into the workpiece?
Thanks

Harvie, 2018-12-17:
Good point! Do you dare to create pull request? :-)

leofaoro, 2018-12-18:
While I know some python, I have never made a contribution. I'd love to learn this.

MARIOBASZ, 2018-12-18:
Look at the file cnc.py, it deals with the tool change in the sector:
code to change manually tool

sfinexer, 2018-12-18:
I think this is the prerogative of G-code and not the program B-cnc

leofaoro, 2018-12-18:
Thanks. I know where to find this. I have to learn the process of using GIT.

leofaoro, 2018-12-19:
> I think this is the prerogative of G-code and not the program B-cnc.
> 
> And if the tool is the filament?
> That at program delay its overheating is possible (for example in pyrography)?

hmm. this is true, but AFAIK bCNC is not for 3D printing, but I may be mistaken. I know using it with laser is also supported. I'll let the maintainers decide. I added a pull request.

sfinexer, 2018-12-20:
> hmm. this is true, but AFAIK bCNC is not for 3D printing, but I may be mistaken. I know using it with laser is also supported. I'll let the maintainers decide. I added a pull request.

B-cnc is used to send G-code, including for 3D models (engraving, burning, printing).

G-code should be sent without any changes.
When creating a G-code, its author (automatic script or otherwise) should take into account all the features of the machine in G-code.

Harvie, 2018-12-20:
I guess the pause can be optional selectable in UI...

[image]

MARIOBASZ, 2018-12-22:
Thanks and congratulations for contributing a plugin. Maybe create a new field to choose waiting time before starting spindle. In the probe / tool change sector or in the naquina configuration.

leofaoro, 2018-12-24:
I modified my version to include an option to set the time for the spindle to wait. I am away from my CNC so I can't test. Can I submit as a pull request and so someone can test before merging? Newbie here so just learning how to collaborate. thanks

hamster65, 2019-01-04:
Hi,
I am also interested in this feature. I do not have an automated tool change but my spindle has a very slow soft start which cannot be switched off and if the position is close to the first "contact coordinates", it is not sufficient time to spin up.

leofaoro, 2019-01-04:
It’s been pulled into master with 5s delay.