vlachoudis/bCNC #1108: B-cnc stops sending data

sfinexer, 2018-12-22:
As can be seen in the video, data is accumulated in the Arduino buffer, although in response to all the data the controller sends the correct response.

http://cnc.tiven.ru/ssr-2018-12-22_18.27.31.mp4

Code arduino:
void setup() {
    Serial.begin(115200);
}
void loop() {}

String inputString = "";  
void serialEvent() 
{
    while (Serial.available()) {
    char inChar = (char)Serial.read();  // get the new byte
    inputString += inChar;
    if (inChar == '\n') {
      //Serial.println(inputString);
      Serial.println("ok");
      inputString = "";
      delay(5);
      }
}
}

Harvie, 2018-12-25:
> I got the impression that the "ok" commands are going too fast, and B-cnc does not have time to accept them, or process them.

Add delay to arduino code.

Does this happen with real GRBL? If not, this issue would be irrelevant.

sfinexer, 2018-12-25:
@Harvie Not. There are many variations of offline grbl on the internet. But grbl itself is not provided for this mode of operation.
There is an idea to remake grbl so that there was an opportunity to work with SD-card. For this, according to the specification, grbl, upon receipt of a command, confirms it with "ok", thereby having to emulate data acquisition.
I implemented it, but the data ceases to be sent for reasons unknown to me.
And in this way, B-cnc will be able to transfer all code at once into a buffer, which will be executed independently of B-cnc.

Harvie, 2018-12-25:
You just suppose that your GRBL protocol implementation is correct. But i think this might not be the case. I think that GRBL is not supposed to respond to `?` command with `ok`. And there can be even more similar cases that we are not aware of.

Harvie, 2018-12-28:
I've just tried to run big job on my machine and it stoped sending after ~45k lines of g-code.
Now trying UGS platform sender, but so far so good...
I guess we have to figure this out...

sfinexer, 2018-12-28:
> I've just tried to run big job on my machine and it stoped sending after ~45k lines of g-code.
> Now trying UGS platform sender, but so far so good...
> I guess we have to figure this out...

I have not yet verified the command using spy.
But be sure to do it.

Harvie, 2018-12-29:
https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#streaming-a-g-code-program-to-grbl
https://github.com/grbl/grbl/blob/master/doc/script/simple_stream.py
https://github.com/grbl/grbl/blob/master/doc/script/stream.py

Harvie, 2018-12-29:
```
$ grep -ri RX_BUFFER_SIZE *py
bCNC.py:RX_BUFFER_SIZE = 128
__main__.py:RX_BUFFER_SIZE = 128
Sender.py:RX_BUFFER_SIZE = 128
Sender.py:		return self._sumcline * 100. / RX_BUFFER_SIZE
Sender.py:			if tosend is not None and sum(cline) < RX_BUFFER_SIZE:
```

Character counting is somewhat prepared in GRBL1.py:
```
elif word[0] == "Bf":
        try:
                CNC.vars["planner"] = int(word[1])
                CNC.vars["rxbytes"] = int(word[2])
```

Harvie, 2018-12-29:
> I've just tried to run big job on my machine and it stoped sending after ~45k lines of g-code.

Problem is that i tried again and i can't reproduce the bug :-/
I am using different arduino, i'll try again once i get back to the same machine.
Maybe it wasn't bCNCs fault.

Also the Sender code is not really documented, so i wasn't really able to figure out any obvious problems after messing with it for hour or two.

hamster65, 2019-01-03:
Hi, 

I am a "CNC beginner" and it seems I have this problem, too. 

- When I try to write settings to the eeprom, bCNC just stops / waits forever while UGS plattform works without any issue. 

- I created a simple gcode with the flatten plugin. It stops after a random time. Example:
*click on start button*
G90
ok
M3S12000
ok
G4P3
ok
G0Z10
ok
g0z3
ok
g0x229.1425y2.8575
ok
g1z0f500
ok
*cnc mill sits there forever*

GRBL still works. I can test this because I have build a hardware solution to "inject" jog and other commands. When I turn the knob on the pendant, the motors move and bCNC displays "JOG" and updates the positions. So communication is working but it is not in a run state and does not send the gCodes.

Harvie, 2019-01-04:
Is that the g-code you have troubles with? or is it longer?

```
G90
M3S12000
G4P3
G0Z10
g0z3
g0x229.1425y2.8575
g1z0f500
```

Harvie, 2019-01-04:
> http://cnc.tiven.ru/ssr-2018-12-22_18.27.31.mp4

@sfinexer why bcnc in the video always says "not connected"? try to fix that.

> data is accumulated in the Arduino buffer

Arduino buffer is not in bCNC. It's in arduino. bCNCs job is not to send any more commands, while the arduino buffer is full. Because arduino has only little memory and if you keep sending the commands, it will simply run out of space, so you only send few commands to arduino buffer and once the arduino finishes them it will have space to store some more commands. Only then you can continue sending more commands. So if there is something left in buffer, it's arduinos fault.

But it's possible that bCNC looses track of what is in arduino buffer.

hamster65, 2019-01-04:
@Harvie 
The gCode was much longer. I am running a couple of tests now. I suspect that a "ok" from GRBL did not arrive at the computer running bCNC but on the other hand, there is an "ok" as last message on the console.

lienbacher, 2019-01-04:
I am experiencing the very same problem. It randomly stops at a random position. Terminal shows last msg as „ok“, status switches to idle but theres still lines in the buffer of bcnc. Never seems to be the same line, so it‘s likely not gcode related.

Harvie, 2019-01-05:
@sfinexer so as i expected, there is problem with your arduino code. I made my own dummy grbl simulator that runs on Linux (no need for arduino):
https://github.com/vlachoudis/bCNC/blob/master/tests/fake-grbl.sh

This is the important code:

`[ "$byte" = '?' ] && echo "<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>"`

Everytime you receive "?" byte, you should send status report. bCNC will not stop streaming until it finds grbl in idle mode.

Harvie, 2019-01-06:
@sfinexer also i've added fixed arduino version:
https://github.com/vlachoudis/bCNC/blob/master/tests/fake-grbl-arduino/fake-grbl-arduino.ino

lienbacher, 2019-01-12:
I do not know about sfinexers issue, but my personal issue was solved by replacing the arduino with a different model (from original reference design to clone with ch340 but dont know if relevant) and the problem ob bCNC stopping sending did not reappear. I also noticed that there was indeed an Ok message missing, but it was further up between two sent lines that there was no OK message in the terminal. It was an empty line instead of an Ok line.

Harvie, 2019-01-13:
Sometimes i feel like there might be problem with EMF or bad contact regarding serial line of arduino (or USB).

sfinexer, 2019-02-16:
@Harvie 

> void setup() {
    Serial.begin(115200);
}
void loop() {}
String inputString = "";  
void serialEvent() 
{
    while (Serial.available()) {
    char inChar = (char)Serial.read();  // get the new byte
    inputString += inChar;
    switch(inChar) {
      case '\n':  Serial.println("ok");
         inputString = "";
         delay(25);
         break;
      case '?':   Serial.println("<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>");
         delay(25);
         break;
     }
   }
}

B-cnc does not finish sending the code.

sfinexer, 2019-02-16:
The code on the SD card does not match the original G-code
> #include <SPI.h>
//#include <SD.h>
#include "SdFat.h"
SdFat SD;
#define SD_CS_PIN 4
File myFile;
float in[128], out[128];
int i;
void setup() {
    Serial.begin(115200);
    Serial.print("Initializing SD card...");
  if (!SD.begin(SD_CS_PIN)) {
    Serial.println("initialization failed!");
    return;
  } 
  myFile = SD.open("test6.txt", FILE_WRITE | FILE_READ); 
  myFile.seek(0);
  Serial.println("initialization done.");
}
void loop() {}
String inputString = "";  
void serialEvent() 
{
    while (Serial.available()) {
    char inChar = (char)Serial.read();  // get the new byte
    switch(inChar) {
      case '\n': {
         Serial.println("ok");
         myFile.println(inputString);
         inputString = "";
         delay(25);
         break;
       }
      case '?':   {
        Serial.println("<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>");
         delay(25);
         break;
        }
      case '!':   {
        myFile.close();
       }
      default :
        inputString += inChar;
     }
   }
}

sfinexer, 2019-02-16:
If you remove the pause, then everything works.

Harvie, 2019-02-17:
I don't get it. How can pause affect this?

sfinexer, 2019-02-17:
B-cnc operation is not terminated.

> void serialEvent() 
{
    while (Serial.available()) {
    inChar = (char)Serial.read();  // get the new byte
    switch(inChar) {
      case '\n': {
         if (inputString>"")
         {
         myFile.println(inputString);
         }
         Serial.println("ok");
         //Serial.flush();
         inputString = "";
        // delay(25);
         break;
       }
      case '?':   {
        Serial.println("<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>");
        //Serial.println("ok");
        // delay(25);
         break;
        }
      case '*':   {
           RUN=1;
           break;
       }
       case 'H':   {
           RUN=2;
           break;
       }
      default :
      {
        inputString += inChar;
       if (inputString=="G90")
       {
        inputString = "";
        start++;
        if (start>1)
        {
        RUN=1;
        myFile.close();
        }
       }
      }
     }
   }
}

sfinexer, 2019-02-18:
@Harvie The problem is that my code works fine with GRBL when it sends (autonomously) commands there.
But for some reason I can not correctly read the data from the B-cnc.
I am making a data buffer for offline GRBL operation.

Harvie, 2019-02-18:
> I am making a data buffer for offline GRBL operation.

such projects already exists...

sfinexer, 2019-02-18:
@Harvie Own development is always better.
I tested it seems to me that B-cnc expects something other than 
<Idle | MPos: 0.000.0.000.0.000 | FS: 0.0 | WCO: 0.000.0.000.0.000>

Harvie, 2019-02-18:
> Own development is always better.

Then why you don't develop your own g-code server (bCNC alternative) from scratch? It's always better to join efforts with other people and reach some point where everyone is happy.

sfinexer, 2019-02-25:
sender.py:
```python
# refresh machine position?
if t-tr > SERIAL_POLL:
	self.serial.write(b"?")
	self.sio_status = True
```
However, this condition is not checked anywhere else.
There is a line:
```python
def parseBracketAngle (self, line, cline):
self.master.sio_status = False
```
I assume that there are not enough conditions for checking the answer.
for example:
```python
# refresh machine position?
if t-tr > SERIAL_POLL:
	if not self.sio_status:
		self.serial.write(b"?")
		self.sio_status = True
		tr = t
```
After all, if the assignment is False/True, then this condition should be used somewhere.
And even so, the question is open, according to the documentation of the GRBL b-CNC should not wait for an answer to the "?" however, if b-CNC does not receive a response, then a buffer overflow seems to occur.

Harvie, 2019-02-25:
> if b-CNC does not receive a response, then a buffer overflow seems to occur.

Do you have any evidence for this? I don't think that status report requests are buffered.

sfinexer, 2019-02-26:
> > if b-CNC does not receive a response, then a buffer overflow seems to occur.
> 
> Do you have any evidence for this? I don't think that status report requests are buffered.

Unfortunately, no ((((((

Harvie, 2019-02-26:
> The cnc code is correct, as the original GRBL accepts and executes it.

What? But we only support original GRBL. What kind of unoriginal GRBL are you using?

> But there just b-CNC stops when sending the code, part of the code is sent, and then everything stops. And the stop is clearly not in a random place.

Maybe try different arduino with good quality short USB cable and nothing connected to it...

I am regulary using git master bCNC to mill and i don't have any issues...

Harvie, 2019-02-26:
I've just moved code responsible for querying status reports to controller plugin: 6fcf00ce6305b6edfc9c29e8bde77e01fe34cd68

sfinexer, 2019-02-26:
> What? But we only support original GRBL. What kind of unoriginal GRBL are you using?

I meant that the cnc code is correct.

> Maybe try different arduino with good quality short USB cable and nothing connected to it...

I tried UNO and Mega2560

> I've just moved code responsible for querying status reports to controller plugin: [6fcf00c](https://github.com/vlachoudis/bCNC/commit/6fcf00ce6305b6edfc9c29e8bde77e01fe34cd68)

I'm going to try now...

Harvie, 2019-02-26:
> I'm going to try now...

I don't think it fixed anything. Just made the code more maintainable...

sfinexer, 2019-02-26:
@Harvie 
Video:
http://cnc.tiven.ru/ssr-2019-02-26_15.36.55.mp4

Harvie, 2019-02-26:
> I download grbl and everything works. This proves that the problem is not in the arduino, as it executes the code perfectly.

There might be a problem with fake_grbl.ino. This means this is not related to this issue. This issue is about bug in bCNC not about bugs in fake_grbl.ino.

On the other hand i would like to get to some resolution of this ticket...
Do you have any evidence (or suspition) that there's a bug in bCNC which causes it to misbehave with proper GRBL instalation? I am milling using git master without any issues, so i can't agree with the claim that **bCNC stops sending data**.

sfinexer, 2019-02-26:
I studied some of the requests from b-CNC.
Added answers to $$ and $ X which b-CNC sends at the very beginning.

However, when connecting to fake_grbl.ino, I see that some of the commands from b-CNC are in the send queue.
And I assume that I miss the answer to a certain request.
But I did not manage to find out what exactly b-CNC expects.

Here is the video, there are blank lines in the b-CNC console.
http://cnc.tiven.ru/ssr-2019-02-26_20.38.16.mp4
[test3.ino.zip](https://github.com/vlachoudis/bCNC/files/2906787/test3.ino.zip)

sfinexer, 2019-02-26:
> Do you have any evidence (or suspition) that there's a bug in bCNC which causes it to misbehave with proper GRBL instalation?

If GRBL is properly installed, I do not observe errors.

Harvie, 2019-02-26:
Duplicate of #933

Harvie, 2019-02-26:
OK i've closed this. If you want to develop fake GRBL arduino sketch, please discuss in #1166

closed 2019-02-26 by Harvie