vlachoudis/bCNC #1118: Another interface for a "better" usability

onekk, 2018-12-27:
I'm modifying the interface to achieve a "better" use of bCNC, in my use case, I'm limited to a small screen as I'm using a SOM that use an old VGA monitor, I'm using an Olimex Board similar in performance to an old raspberry Pi.

I have tweaked the interface to obtain some behaviour:

1) a separate Zstep settings (already implemented in bCNC by me some time ago)
2) some preset (that could be changed in the .bCNC ini file )
3) a fine button that use a different (and smaller step ) when the "fine" button is pressed

I'm working also in a small CNC panel, for some "most used" machining operations.
the goal is to obtain some button maybe a "mem A", "mem B" , "z start" and "z end" and some others buttons, like "line" "plane" to obtain simple functions like:

1) move the gantry to the first point and press "mem A"
2) move the gantry to the last point and press "mem B"
3) set the "z start" and memorize the Z height (on the stock piece)
4) input the "z end" or move the gantry to the CNC plane that will became final Z and press "Z end"
5) select "line" and the CNC will mill a line with a given stepover till the Z end is reached.

All this work has to be done in the "Control" screen like some old CNC control panels

the plane woul use the A and B as rectangle corners and mill "this pocket" as when you face a stock piece, to obtain a milled plane, or when "surfacing" the holding plane on your CNC.

As I'm using a Shapeoko I think that it is a more "natural way" than to use the Tools (page)

But I'm not very skilled in the bCNC internals so some help is required, and more discussion are welcomed.

Here a little screenshot of the actual work. 

[image]

This interface is fairly "tested" in the real use by me, but some modification are "very new" as I've adpated some old code to a freshly downloaded bCNC codebase.

onekk, 2018-12-27:
The interface may look confusing, but some suggestion are welcomed.

For now I'm searching a way to obtain current X Y Z positions, to achieve the "mem A" and "mem B" functions.

Some suggestions?

Harvie, 2018-12-27:
>  the "mem A" and "mem B" functions.

There are already two features to achieve that in bCNC and GRBL.
GRBL has multiple work offsets
bCNC has digitizer #929 

No need to reinvent wheel.

onekk, 2018-12-27:
Not wanting to reinvent the wheel, maybe adding something to it.

The working coordinates are one thing, the A and B position are another.

I want to be able to:
1)  jog to a position, memorize it (maybe only the X and Y position)
2) jog to another position, memorize it (same thing for the coordinates)
3) set the workpiece zero
4) mill a "line" from A to B with the proper stepdown (maybe specifing in a dialog the target depth and the tool diameter)

Is this possible to achieve with the actual bCNC  settings.

The documentation of the CAM features seem to lacks some informations.

Regards
Carlo D.

Harvie, 2018-12-27:
@onekk Yes this is possible in latest git master. I've sent you link to #929
There is not much documentation. I will surely make video about it once 0.9.15 get officialy released.

onekk, 2018-12-28:
Ok I've worked some more on defining the things in the interface, I've simplified some more the interface, and added some quick recalls.

So the interface look more clean
[image]

The crowded lines are out:
1) the first combobox in the XY blocks of control are for the step1 preset that is used by the + and - button to increment or decrement the step without having the power behaviour, so if you set 1 mm as step1 you can use the + or - button to increment or decrement the value by the step1 so if you arrive at 10 instead of passing to 20 you pass to 11 and so on.
2) the second and third buttons are configurable using the right click that open a tkSimpleDialog.askfloat that permit to change the value in a more clean manner

3) the memA and mem B and the line are working, but some work is needed as they use the machine coordinate system 

I've some question but are for another time.

Regards

Carlo D.

onekk, 2018-12-29:
I don't ask for a merge or want to impose my view, simply let the people that have expressed in the past some interest in an alternative interface discuss another way to the "usual" things with more speed and less fiddling around with the mouse.

This discussion is only for collecting some ideas (I haven't made a proper pull request as the code is in active development).

Some time ago there were some interest in a smaller interface for the embedded processors like Raspberry Pi, but I haven't had times to implement the features.

Having some control on the "main page" is very handy, not much than a line and a flatten command, some other exotic works are IMHO to be done by a proper CAM package.

Having the  ability to use some "preset" step are handy as they are implemented in the KP1, KP2 and KP3 key press, having them on the main interface seem more "logical".

As the separate Z and XY step settings, they are used when you have to change the work piece and then find the "zero", having one step say 10mm and change it to 0.1 mm as is usually done when using the "paper sheet" technique is very handy and also having a 5 or 10 mm step is handy when you want to raise the Z axis to have enough clearance when jogging past the fixtures.

The separate z step was implemented in the early days by me and adopted as option by @vlachoudis 

As in the jogging of XY a fine tune of 1mm or even much smaller is handy but even a 50 or 100 mm is handy to have far from a button click than to select a value on a listbox control.

But as usual is a matter of taste, and the different tastes make the life better.

Regards
Carlo D.

PCSDIAS, 2019-01-17:
@onekk I think there is an easier way to get what you want and much more.  Leave the bCNC interface alone and get yourself a web pendant to fulfill all your needs.  I was not happy with the original bCNC Web pendant and decided to make my own to run on an old cellphone.  It does everyrhig I need, but loading the Gcode file, because the phone does not have the USB OTG capability.  The bCNC http server offers a lot of resources.  The results are [here](https://youtu.be/YrxuQzEnp40).  The video only shows the jogging implementation, but I suppose it will give you an idea of what is possible.  The pendant html, js, css and json files sum 27KB.  If the graphic files are considered the whole package goes to less than 60KB.   
Paulo

Harvie, 2019-01-18:
> For the pendant, i prefer a wired one, too many RF interferences around the machine.

1.) You can use web pendant localy on the same machine as bCNC
2.) I hope you will fix your RF interferences soon :-) There are always people struck by unreproducible RF problems filling github issues wondering why their GRBL and bCNC does not work. Have good luck with your Z axis anyway :)

PCSDIAS, 2019-01-19:
@Harvie @onekk My GRBL board is connected to the bCNC running laptop via Bluetooth.  The web pendant is an old cellphone.  The cellphone wifi router is enabled, so there is no need of a external wifi router.  No evidence of EMI, no bad connections and no tangled cables lying around.
The only problem I had was spindler interfering on the limit switches.  It was solved with three 100nF capacitors soldered directly on the microprocessor input and ground pins.
Since then, it has been working for a year without any fault.
Paulo.

onekk, 2019-01-27:
The Z axis is fixed, and in the meantime I've used the keyboard as a pendant, (maybe is too big for the scope, but has simplified my work on setting the Z axis.

In the meantime, I've worked on some "improvement" on the "Machine functions" on the Control panel, that I've found very handy.

I've put two buttons, memA and memB, ad another 2 buttons "ret A" and "ret B" to go to the previous memorized position (A and B).
My machine is not very sturdy so  from time to time I've the need to verify a well known position to check if something in the machine are come loose so after a "Home" a check to the opposite corner or maye at the zero of the fixture is very handy to have on the "Control Page".

This is very similar in concept with some functions already present in bCNC like the digitizer, but I've not found already explanation on how it works.

A quick look at the code make me suspect that many position could be memorized, but with so little information I couldn't be sure. Some hints?  

It will be a good thing to have it in the "control" page or maybe integrate them with some macro in the "user buttons".

Another functions That I have "done by hand"  using standard gcode, but I suspect it is already in bCNC is the correction of the coordinates useful when using a "laser pointer".

I.e. I put the laser "crossbeam" on the milling point and then move the tool on this point and then zero the XY coordinates.

I suspect that is already implemented for the camera, maybe already present even in the "macro" language. (A side note there is some documentation about the macro that could be used in the "user buttons").

Not my intention to complicate the life of anyone, but only some "real work" problems to solve, and I suspect that this problems are not only mine, but other people could benefit of these functions.

I intend to be "costructive" and not "destructive" about bCNC (I fell like a sort of early adopter, and whenever asked I suggest to use bCNC in pair with GRBL as a good choice for hobby CNC, and even for small shops).

TIA and regards.

Carlo D.

onekk, 2019-11-27:
Some work have been done here a little screenshot

[image]