vlachoudis/bCNC #1119: Can maximum Z-travel be limited relative to maximum z-height?

JanneKro, 2018-12-27:
(version: rpi - 0.9.14)

I'm using an X-Carve and I can use the home-button to all axis stop and finetune their positions when hitting the limit-switches in the down-left corner.
I would imagine that that this should make bCNC not allow setting a z-position higher than this limit, but at least for me, this is not the case.
I'm working with thick material with little to spare (maybe 10mm from tool-tip to material), and when I start the job, It seems like this particular g-code wants to go above 10mm of the material initially, which screws everything up. Is this supposed to be the case? 

It seems like maybe i haven't understood if "homing" does anything useful at all (I reset XYZ according to the material and expected homing to do something. But I've also seen that it doesn't stop med from jogging past borders either, resulting in recalibration of the positions with respect to the material).

Is this as intended, or is there an issue with registering my limit-switches outside of homing?

Harvie, 2018-12-27:
You can set soft limits in GRBL, that way GRBL will automaticaly fail if something tries to go further than enabled.
If bCNC generated g-code tries to go further than that, you can lower safe-z.
If you are using other software to generate g-code, then you are on your own.

closed 2018-12-27 by Harvie

JanneKro, 2018-12-27:
Thanks for the response, @Harvie.
Yes, I just realized before I saw your answer that the Soft Limits were not enabled in GRBL.
I did try again with them enabled and bCNC stopped immediately with ALARM:2.  So my final question regarding this is whether bCNC checks the whole job before running, or (more likely) the first move was simply to raise the tool too far.

I generate my gcode with VCarve. As an idea/suggestion, It would be nice if bCNC had a "virtual" z-level for instances like this.  When one sets up the job and uses a specific tool, it can be obvious that one is "in a pinch", but as long as one is still above the material and has ensured that the are no obstructions, it is safe to run the job, and any z-position above the limit is merely handled virtually. That is, the height is capped at Z-max.