vlachoudis/bCNC #1124: Code don't adhere to the PEP8 python specifications

onekk, 2019-01-02:


Harvie, 2019-01-03:
I prefer tabs and my favourite editor doesn't work very well with spaces. sorry.
I don't want to discus this right now, because this can only lead to holy flamewar without any actual benefit to bCNC project.

But recently i've fixed all files, where there was mixed use of tabs and spaces.
Also i've added travis test that will check all pull requests for mixing of tabs and spaces.

If you need this because of your editor, you can setup "git smudge/clean" to automaticaly replace this on checkout and commit.

closed 2019-01-03 by Harvie

oddstr13, 2019-01-14:
May I ask what your editor of choice is?
Most modern editors can be configured to use 4*space as if it where tab.

Harvie, 2019-01-14:
Another step of this discussion: Why even use python, when you can use XXXXXXX....