vlachoudis/bCNC #1125: Error In Pendant.py

qvisionsa, 2019-01-03:
I think there is an error in line 231
it is written
`f = open(os.path.join(webpath,page),"r")` 
this way the web server can't open image files (jpg, png etc)
I thin the correct is
`f = open(os.path.join(webpath,page),"rb")`

Harvie, 2019-01-03:
Have you tried if that fixes your problem?
Isn't that affecting loading of text files like .html, .css or .js?

If you've tested this, can you please make pull request with this modification, so i can merge it?

qvisionsa, 2019-01-03:
This change fixed my problem instead. It seems that isn't affecting other file types, as I'm loading CSS, js, png HTML without a problem. 
Anyway, I will test it thoroughly and I will let you know. For the moment it's fine!