vlachoudis/bCNC #1126: Swap X and Y axis buttons

ralfsteck, 2019-01-03:
Hi all,

I woulf like to have an option to swap the X and Y axis buttons in the Jog Display.

The reason for this is that my machine is 800mm long in the direction the gantry moves (which is X by Shapeoko conventions) and 500 in the direcztion the Z axis moves on the gantry (which is Y by Shapeoko conventions). I have placed my machine with the long side at the wall and sit on the other long side. So the X+/- buttons, which are displayed "up" and "down" on the monitor actally move the gantry left/right, The Y+/- buttons (which are located "left" and "right" in nthe display) move the Z axis to/way from me. This makes it very difficult to jog intuitively.

I have switched the axes on my machine, but I think it would be more ligical to hve an option to move the buttons in the display. Now all the motor driver labels are wrong etc.

Harvie, 2019-01-04:
> This makes it very difficult to jog intuitively.

Every CNC operator should train his spatial intelligence :-) I jog from various angles all the time. And yes, sometimes i screw :-D Also note that i jog using wireless numeric keyboard, which makes this so much easier, since you can carry it around and rotate it anyway you want. It's well supported in bCNC and they are very cheap.

[image]

> I would like to have an option to swap the X and Y axis buttons in the Jog Display.

I have some plans to make UI much more configurable, but i guess it will take some time, because there are much more important tasks right now.

> I have switched the axes on my machine, but I think it would be more ligical to hve an option to move the buttons in the display. Now all the motor driver labels are wrong etc.

In fact this is exactly what i would do in your place. If the labels are wrong, you can recompile GRBL with different stepper gpio pin numbers. or relabel :-)

I don't say your point is completely wrong, but implementing this and maintaining such code would take some time which we don't have right now. But i will keep this in mind and reflect this, once there will be major UI rewrite.

closed 2019-01-04 by Harvie

ralfsteck, 2019-01-04:
If this is adressed in a more configurable UI some time in the future, I'm perfectly happy. Thanks for all the work on this wonderful piece of software.