vlachoudis/bCNC #1127: Autolevel error grid

luiziko014, 2019-01-04:
I'm doing the autolevel, I have the following problem. When I try to use Auto Level function , the machine run normally but the graphic in the bcnc no draw the correct location , all are made in the corner near the origin.

Thanks soo much for your reply , but the problem continues, I installed bCNC using PIP , I tried using other versions without success. Please help me!
Requirement already up-to-date: bCNC in c:\python27\lib\site-packages
Requirement already up-to-date: pyserial<=3.0.1 in c:\python27\lib\site-packages (from bCNC)
Requirement already up-to-date: numpy>=1.12 in c:\python27\lib\site-packages (from bCNC)
Requirement already up-to-date: Pillow>=4.0 in c:\python27\lib\site-packages (from bCNC)

[image]
[image]

Harvie, 2019-01-04:
What version of grbl do you use?

luiziko014, 2019-01-04:
> What version of grbl do you use?

I use 0.9i grbl version, I tried with others version, olders and newers but the Bcnc do the same

Harvie, 2019-01-04:
Please try install opencv-python>=2.4 using pip and use grbl 1.1f, Legacy support of GRBL 0.x is not currently maintained very well in bCNC. Grbl v1 is handled by completely different code that Grbl v0 in bCNC, the legacy v0 might be buggy.

If that helps, please tell me which of these two fixed your problem.

luiziko014, 2019-01-04:
> Please try install opencv-python>=2.4 using pip and use grbl 1.1f, Legacy support of GRBL 0.x is not currently maintained very well in bCNC. Grbl v1 is handled by completely different code that Grbl [...]

Thanksssss soo much. I solve my problem with your help. 

I upgrade the grbl 1.1f and I began to carefully review each of the parameters. I had the units in inches and also one of the masks was misplaced (to take as reference the version grbl 0.8).
With all that resolved, it was not necessary to install the opencv python.

Thank you very much, greetings from Peru
[image]