vlachoudis/bCNC #1132: Camera probe doubts

beckmx, 2019-01-11:
Sorry for opening an issue, I had some doubts, I have this camera and I am going to adapt it, I was wondering, if this camera probe looks and searches for the milling "probe" and aligns itself. I was going to make that feature but I see this, I read the wiki also but doesn't really says what it does, I could happily also do a video, just to make it clearer for newbies

Harvie, 2019-01-11:
https://github.com/vlachoudis/bCNC/wiki/Probe-Camera-Alignment

[image]

You just manualy calibrate your camera. bCNC has to know how far from the spindle the camera is.
Once you have this calibration done, you manualy jog until you align the crosshair with point on your work table where you want to be and click the button to switch from camera to spindle. This will cause the spindle to move to the exact point that you previously saw on camera aiming cross.

Also if you use latest development version of bCNC, you can use camera to "record" points and lines to measure or make drawing of part using camera.

There is no computer vision that automaticaly finds stuff. You just click jog buttons until you move your camera crosshair where you want it to be.

closed 2019-01-11 by Harvie

onekk, 2019-02-02:
This behaviour could be adpated to a laser pointer.

I've achieved such a similar work using the user buttons, but it will be handy to have implemented better than sending "manual" sommand, maybe some .bCNC ini variables holding the XY offset of the laser pointer and two button on  the control page.

Laser to Tool position (L2T)
Tool to Laser position (T2L)
Regards

Carlo D.