vlachoudis/bCNC #1133: bCNC always says "not connected"

roamerP, 2019-01-13:
I have problem on Windows 10 x64. bCNC always says "not connected" in red. But it communicates with arduino. I am not sure, what is the problem.

This is the error I get in command prompt:

Traceback (most recent call last):
File "D:\bCNC-master\ToolsPage.py", line 973, in init
exec("import %s"%(name))
File "", line 1, in 
File "D:\bCNC-master\plugins\slicemesh.py", line 32, in 
import stl #FIXME: write smaller STL parser
File "D:\bCNC-master\lib\stl_init_.py", line 1, in 
from .stl import BUFFER_SIZE
File "D:\bCNC-master\lib\stl\stl.py", line 12, in 
from . import base
File "D:\bCNC-master\lib\stl\base.py", line 21, in 
class Dimension(enum.IntEnum):
AttributeError: 'module' object has no attribute 'IntEnum'

closed 2019-01-13 by roamerP

Harvie, 2019-01-13:
Hi. Why did you closed this issue? Have you find the solution?

roamerP, 2019-01-16:
Yes, in post #544 vlachoudis asked if they can use command $10=1 and after that bCNC works again.

ps, sorry for late replay