vlachoudis/bCNC #1134: Question 4th axis

walterwissmann, 2019-01-13:
Hello to all,

how is the status of the fourth axis?

Regards
Walter

Harvie, 2019-01-13:
No such code was ever merged to bCNC. If somebody develops this, i am willing to merge such feature under following circumstances:

* It will not break support of 3-axis controllers
* The GUI will be non-obtrusive (no need to hide it on 3-axis machines, just make it compact)
* The GUI will be prepared to support up to 6 axes in future

Also what kind of controller do you want to use for additional axes? As far as i know the stock GRBL does not support this...

For more see this label: https://github.com/vlachoudis/bCNC/labels/more%20than%203%20axes

closed 2019-01-13 by Harvie

MARIOBASZ, 2019-01-16:
As I read, G2Core runs 5 axes, I think 6 also (look Pocket NC) coordinated. You can add more axes but with latency drawbacks to coordinate movements

Harvie, 2019-01-16:
Yes, tinyg / g2core support is requested for very long time: #83

dylanpil, 2020-06-14:
Hi Harvie,
I've used this for fork for extra axis.
https://github.com/dguerizec/bCNC-4axis
It would be great if you could add the 4th axis in your version. :)

Harvie, 2020-06-14:
6 axis support was already added in #1384

Harvie, 2020-06-24:
But there seems to be bug: #1426