vlachoudis/bCNC #1135: how to display current positions of axes in gcode with %python

coppella, 2019-01-14:
Hello, I hope someone can help me. I'm a beginner in Python and can't get any further. I have integrated a GCode with % line python code, it all works so far. My problem is that I need the current positions of the axes in my code, how can I save the positions in a variable? then I would like to move to a new position with %line  python script, is that possible ?

Harvie, 2019-01-14:
Following variables are available as seen in `CNC.py`. I think you are interested in `wx`,`wy` and `wz`. But note that these might not be up to date at the time the command is issued, because machine is moving or didn't reported its position yet.

```python
        vars           = {
                        "prbx"       : 0.0,
                        "prby"       : 0.0,
                        "prbz"       : 0.0,
                        "prbcmd"     : "G38.2",
                        "prbfeed"    : 10.,
                        "errline"    : "",
                        "wx"         : 0.0,
                        "wy"         : 0.0,
                        "wz"         : 0.0,
                        "mx"         : 0.0,
                        "my"         : 0.0,
                        "mz"         : 0.0,
                        "wcox"       : 0.0,
                        "wcoy"       : 0.0,
                        "wcoz"       : 0.0,
                        "curfeed"    : 0.0,
                        "curspindle" : 0.0,
                        "_camwx"     : 0.0,
                        "_camwy"     : 0.0,
                        "G"          : [],
                        "TLO"        : 0.0,
                        "motion"     : "G0",
                        "WCS"        : "G54",
                        "plane"      : "G17",
                        "feedmode"   : "G94",
                        "distance"   : "G90",
                        "arc"        : "G91.1",
                        "units"      : "G20",
                        "cutter"     : "",
                        "tlo"        : "",
                        "program"    : "M0",
                        "spindle"    : "M5",
                        "coolant"    : "M9",
                        "tool"       : 0,
                        "feed"       : 0.0,
                        "rpm"        : 0.0,

                        "planner"    : 0,
                        "rxbytes"    : 0,

                        "OvFeed"     : 100,     # Override status
                        "OvRapid"    : 100,
                        "OvSpindle"  : 100,
                        "_OvChanged" : False,
                        "_OvFeed"    : 100,     # Override target values
                        "_OvRapid"   : 100,
                        "_OvSpindle" : 100,

                        "diameter"   : 3.175,   # Tool diameter
                        "cutfeed"    : 1000.,   # Material feed for cutting
                        "cutfeedz"   : 500.,    # Material feed for cutting
                        "safe"       : 3.,
                        "state"      : "",
                        "pins"       : "",
                        "msg"        : "",
                        "stepz"      : 1.,
                        "surface"    : 0.,
                        "thickness"  : 5.,
                        "stepover"   : 40.,

                        "PRB"        : None,
                        "TLO"        : 0.,

                        "version"    : "",
                        "controller" : "",
                        "running"    : False,
                }
```

You can access these variables using square brackets in g-code: `G90G0Z[safe]` or directly in python calls: `%print(safe)`

Harvie, 2019-01-14:
Also if you tell me what exactly you are trying to achieve, i can suggest you some solution.

closed 2019-01-14 by Harvie