vlachoudis/bCNC #1137: BLTouch probing and leveling with bCNC

CCSnell, 2019-01-16:
OK, so my setup is a bit (over)complicated. I have added a [BLTouch probe](https://www.antclabs.com/bltouch) to my [RPI-CNC](https://wiki.protoneer.co.nz/Raspberry_Pi_CNC#Version_2.51) with a Arduino Nano running an unmodified (from source code) GRBL v1.1f (I use the board without the RPi and just connect over USB). I use this as a controller for a modified K40 Laser Engraver with an adjustable bed. After trying to find a way to utilize the available remaining pins for the probe (it needs a PWM capable pin to deploy/retract the probe) and not finding an answer I came up with the following low cost solution. I use an [Arduino sketch](https://github.com/CCSnell/BLTouch4bCNC) on a separate Nano that watches for a signal coming from the RPI-CNC Coolant Enable Pin. That part works great. Sending M8 retracts the probe, and M9 deploys and waits for contact, bCNC acknowledges the probe (connected to the Probe pin) impact upon touch. 

What I am trying to accomplish is : 

**- Make the probe retract when homing** 
(Modified Sender.py - Works Great)
`elif cmd == "HOME":`
			`self.sendGCode("M8")`
			`self.home()`

**-Deploy the probe (using M9) just before any probing operation** 
(Works using a script, but needs to be inserted automatically for auto-leveling)
``M9
G38.2Z0F50
%wait
M8``

**-Perform "Move Off/Safe Z height"  AND/OR M8 command just after probing/auto-leveling** 
(Not Working)

How can I get bCNC to insert the proper M8/M9 commands before/after G38.2 or between moves during Auto-leveling?

Terminal Output:
HOMING
> ok
M8
[GC:G38.2 G54 G17 G21 G90 G94 M5 M8 T0 F50 S0]
ok
$G
ok
$H
ok

MOVE TO PROBING AREA
>G0 F200 X30 Y50
[GC:G0 G54 G17 G21 G90 G94 M5 M8 T0 F200 S0]
ok
$G
ok

HOMING SCRIPT
>M9
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F200 S0]
ok
$G
[PRB:-297.999,-165.002,-18.405:1]
ok
G38.2 Z0 F50
ok
M8
[GC:G38.2 G54 G17 G21 G90 G94 M5 M8 T0 F50 S0]
ok

Harvie, 2019-01-16:
> Make the probe retract when homing

Use configurable buttons  to make extra button for this. Rather than modifying bCNC code:

[image]

CCSnell, 2019-01-16:
Thanks, I can do that for the homing (configurable button), and for a (G38.2) single probe operation(configurable button). Is it possible to use auto-leveling through a configurable button where M8/M9 can be issued before and after each probe point, or is there another way?

Harvie, 2019-01-16:
> M8/M9 can be issued before and after each probe point

Why don't you just keep probe deployed during whole autoleveling cycle? eg.:

1.) manualy deploy using button
2.) probe 100 points as usual
3.) retract using another button

Harvie, 2019-01-16:
I plan to do big changes in probing/autoleveling. If i do so, i will remember to add some option to attach custom prefix and suffix g-code to such operation. But this is rather a long term goal.