vlachoudis/bCNC #1142: Direction issues

usajet, 2019-01-19:
When I home it goes to the right rear of the cnc. I go to the corner of the work piece and set my zero. But when I do a scan it wants to go to my homing postition and keep going. Any ideas?

Harvie, 2019-01-20:
What do you mean by "scan"?

usajet, 2019-01-20:
A trial run of pattern that's going to be milled.

Harvie, 2019-01-23:
I am not really sure what do you mean.
I don't have homing switches installed, but i believe that scan should happen in work coordinate system rather than in machine coordinates, so it should not really matter if your homing is done right as long as you zero your WCS properly. Can you make some screenshots or video of what exactly you do when that happens?

closed 2019-01-23 by usajet

Harvie, 2019-01-23:
Is everything OK now?

usajet, 2019-01-25:
Yes thank you

Harvie, 2019-01-26:
Just for the record... Do you mind to share what the problem was? In case that somebody experiences the same?

usajet, 2019-01-26:
It was because of where I  set my zero.