vlachoudis/bCNC #1143: Z motor lock during homing

BernardG, 2019-01-23:
This is quite strange. I must say at first that it has been working fine for a while. Now, if I want to jog the Z axis, either by clicking on the arrows or with the keyboard, I hear the motor emitting a "lock" sound, with a buzz depending on the distance selected, but it does not move. Other axis works fine.

I did my best to see where the problem was coming from, changing driver, cable, motor, even the CNC Shield, but it looks like it really comes from bCNC, as it's working fine with UGS Platform. Ho, I was going to forget, it does the same on Ubuntu Linux and Windows 10.

Is there a setting somewhere which would limit Z axis jogging?

MARIOBASZ, 2019-01-23:
If it's not a problem with a loose cable, does it do the same going up or going down? The problem is with both G0 and G1 at low speed?
controls driver temperature
if it's just going up, maybe it's high speed or acceleration

BernardG, 2019-01-23:
> If it's not a problem with a loose wire, does it do the same going up or going down? The problem is with both G0 and G1 at low speed?

Any speed, it's the same,and both ways, up and down

> controls driver temperature

I did, even added a fan, same...

> if it's just going up, maybe it's high speed or acceleration

Nope, both ways. To make sure and test, I dismantled the motor from the machine, to be sure it was away from the limit switches cables.

And, to repeat myself, it works with another software....

BernardG, 2019-01-23:
Well, I believe I must apologize for this false alarm! Took me the most of 2 full days to realize this is due to a random false contact on my extension cable between the motor and the shield. So when I touched something it was working, then touching again, and not working anymore. This is what led me to believe it was happening only with bCNC!

Further tests showed that it was not working "sometimes" with GCS, too. So, I will change that cable, redo the wiring, and hope it definitely end that ennoying recurring problem.

closed 2019-01-23 by BernardG

Harvie, 2019-01-23:
See #1144