vlachoudis/bCNC #1147: xbox jogging on raspi and GBRL 1.1 Shapeoko 3 XXL (v.2.4)

aggiedan977, 2019-01-26:
I have search for related post and found some info on xbox controller jogging but nothing definitive on how to set it up. I have attempted to load the xboxdrv and joystick with no luck. I have read about setting up shortcuts but those look like keystroke shortcuts. I can't determine if there is a joystick wedge app that should interpret input as keystrokes and thereby execute bCNC shortcut keystroke mappings.

Any guidance here would be appreciated.

aggiedan977, 2019-01-26:
I've also read about Xpadder but am hesitant to shell out $$ until I understand how it should be configured.

onekk, 2019-02-02:
I'm interested in using one of these but wired.

When in numeric keypad i could use the arrow button to jog the X and Y and the "PgUP" and "PgDn" to control the Z axis.

plus + and - to change the "jogging step" and event the * and \ to use the multiply function.

But I'm stuck with the absence (without pressing the Num Lock button) of a manner to select the steps.

I often use a setting of 0.5 mm and a more coarse one say 10 mm.

It will be very handy to select them with one keystroke, maybe using a key to toggle between step1 and step2.

How this could be achieved?

Harvie, 2019-02-03:
I use * and / buttons to adjusts the step...

onekk, 2019-02-28:
Yes I have find the / * buttons handy, but a predefined step button are way better, speaking of operations on the machine.

You often need a fine tuning 0,5 or even 0,1 mm and a more coarse jogging like 5 or 10 mm.

so 5 / 5 = 1mm and 10 /5 is 2mm 2/5 is 0.4.

Having a button on the keypad that cycle through the 3 step step1 step2, step3, step1 ....  i think is more handy and useful i could put 0.1 0.5 10  in step step2 and step3 and toggle with one button.

Harvie, 2019-03-04:
I've just randomly came across following code, which seems like there's already such feature:

bCNC.ini:
```
step1  = 0.1
step2  = 1
step3  = 10
```

bCNC.py:
```python
                self.bind('<Key-1>',            self.control.setStep1)
                self.bind('<Key-2>',            self.control.setStep2)
                self.bind('<Key-3>',            self.control.setStep3)
```

But i never tried it...

onekk, 2019-03-11:
Ok, I have to fiddle with some settings, or eventually change the + and - button to toggle between the steps, or maybe the / * that should be better suited for a keypad as a pendant use, the limited number of keys and the "dual mode" with and without the "bloc num" need a good thinking on a better use to minimize the clicks and the errors during the operation, (I have broken some tool having not noted the amount of steps during some "fine tuning").

I will investigate the code and maybe make some suggestion.

Regards

aggiedan977, 2019-04-12:
I'm really excited to update this thread as I have found a solution. I have been able to use a USB XBox360 wired controller to job my Shapeoko 3XXL with bCNC. There are two additional apps required XBOXDRV and QJOYPAD. More information can be found in the link below. My only issue is a permissions issue where I must execute each app command with 'sudo..' prefix. (Note I start with the XBOXDRX command in one terminal and start QJOYPAD in a second terminal window.

https://openbuilds.com/projectresources/raspberry-pi-xbox-controller-bcnc-and-you.259/?fbclid=IwAR2oY1bv8XQg2VGi7Z8V1EHKDEdwgOdE6ESZ2Pl2pBjDSSvlIKeZ7wuoAhQ

The buttons and joysticks are really easy to program.

Mungbeanz, 2020-01-10:
@Harvie 
do you think it is possible to add in a 4th step on key-4 without destroying some other core function.
what else are keypress 1-5 doing?

> I've just randomly came across following code, which seems like there's already such feature (for >keyboards):
>
> bCNC.ini:
>
>                 step1  = 0.1
>                 step2  = 1
>                 step3  = 10
> bCNC.py:
> 
>                self.bind('<Key-1>',            self.control.setStep1)
>                self.bind('<Key-2>',            self.control.setStep2)
>                self.bind('<Key-3>',            self.control.setStep3)
>
> But i never tried 
>
> it...

Something like:

bCNC.ini:

                step1  = 0.01
                step2  = 0.1
                step3  = 1
                step4  = 10

bCNC.py:

                self.bind('<Key-1>',            self.control.setStep1)
                self.bind('<Key-2>',            self.control.setStep2)
                self.bind('<Key-3>',            self.control.setStep3)
                self.bind('<Key-4>',            self.control.setStep4)