vlachoudis/bCNC #1148: Option to change color for cross and circles in camera

MARIOBASZ, 2019-01-26:
I give a lot of use to the camera, it really helps me. A good improvement would be a good option to change the cross color and the camera circles if the piece is white. Another agile option would be to have a black intermediate circle between the two white circles. Someone interested in implementing it?

Harvie, 2019-01-27:
Maybe just invert the color of camera?

MARIOBASZ, 2019-01-27:
simply do that with a button would be excellent

Harvie, 2019-01-29:
Maybe it can be done realtime. I was thinking about something like this:

[image]

MARIOBASZ, 2019-01-29:
I think it's a very good idea, I do not know how difficult it is to implement.
If there are many lines to visualize on canvas (as happens when I generate a trochoidal cut) my machine becomes very slow (I do not know if it is my laptop or python / bCNC problem). Your idea may aggravate the slowness?