vlachoudis/bCNC #1151: B-cnc stops meeting "!"

sfinexer, 2019-01-31:
B-cnc stops meeting "!" at the beginning of the line. Errors from GRBL no.

Harvie, 2019-02-04:
Why do you put "!" in your g-code?

sfinexer, 2019-02-11:
> Why do you put "!" in your g-code?

Is it really prohibited by the documentation on the G-code?

Harvie, 2019-02-11:
"!" has special meaning in GRBL. You surely can't use it as a comment.

Please stop guessing and read the documentation for linuxcnc g-code and GRBL before creating issues. No you can't just say "im gonna use this g-code because it looks cool". You have to use proper g-code syntax. If you want to use something that is not proper g-code, then you obviously can't use unmodified g-code based tools.

closed 2019-02-11 by Harvie

sfinexer, 2019-02-11:
> "!" has special meaning in GRBL.

What is the importance of GRBL?
I can connect a different controller ...
This code "!" not inconsistent with G-code syntax.

Harvie, 2019-02-11:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

sfinexer, 2019-02-12:
I was wrong.
This is all because in the B-cnc console, I do not see this command in the list of transmitted ones, but only in the buffer.
If I had seen this command in the B-cnc console, I would have no questions!

And without seeing the commands in the list, I assume that there is an error in the B-cnc program.