vlachoudis/bCNC #1152: Autolevel does not work

sjors321, 2019-01-31:
Hello,
Is this issue related to hardware or are there more people with this problem?
Could someone please shine a light on this?
The thing is when i click "scan" he bit retracts before touching the surface.. 
Sonde probing did work before but does the same thing now..
I didn't change the hardware.
How could this be?

Thanks in advance!

sjors321, 2019-01-31:
Hi,
I found out what the problem is.. maybe other people might benefit from this.
I just unplugged my laptop so im only using battery power and that solved it.
Great! Good thing ive got a decent battery ;)

leofaoro, 2019-01-31:
Sounds like a grounding issue?

sjors321, 2019-01-31:
Yeah but how can that be because i only use powersupplies  so everything is isolated i think..
But perhaps its because my powerbrick from the laptop routes the mains ground to my laptop.
But then im thinking someone who's using just a desktop machine would have the same problem..
Any idea?

sjors321, 2019-01-31:
"ground" should be earth...

sjors321, 2019-02-01:
> "ground" should be earth...

Sorry my powerbrick is connected true a non-earthed outlet.

onekk, 2019-02-02:
Many problems will arise from laptop "power brick", in some configurations, the "ground" is not a proper ground and I've fried my Atmega 2560 used in a 3d printer with some power loops using the laptop power brick (it was and old Asus eee pc), there are some cable provide some isolated isolated the usb to avoid such loops but they are expensive.

A "common" solution is to use a Raspberry Pi to make a dedicated controller for the CNC, I use a similar solution using an Olimex card Olinuxino A20 very similar to the Raspberry Pi (It has an A20 Dual core ARM processor), an old monitor (My board could use a VGA output), and a keyboard and mouse will solve every problems, it could be powered with the same DC supply of the GRBL board, (There are some solution with a shield for the Raspberry Pi with an ATmega 328 running GRBL), but usually it is more cheap to use separate boards, and even maybe using the serial output of the Raspberry Pi to connect the serial input of the Atmega 328 jumping the USB connection (but with some board you have to use a 3,3v to 5V adapter for the serial port)

sjors321, 2019-02-02:
Thanks for the extended answer. Its good to know because on my work the want to purchase a cnc machine intended to be used with a laptop.
Is this problem only with older laptops?