vlachoudis/bCNC #1154: Request: How to load a file in the interface.

onekk, 2019-02-02:
The problem.

I want to be able to load a Gcode file in the interface, and executed it.

The scenery is useful even in plugin writing.

Suppose you have a rather complex operation to achieve and you have a mean to:

1) send some position to and external program
2) execute this program
3) load the Gcode file in the editor
4) execute it (using the standard buttons on the control page)

To be able to do so you need to:

1) a way to execute and external program maybe pressing a button in bCNC
2) a way to load a file in the editor and execute it.

1) A "macro" that could be used in the user button to launch and external program (Maybe it could be achieved using python commands in the user button definitions?) 

2) is already done in bCNC, but I haven't achieved the correct steps to call it from a function

TIA and Reagrds.

Carlo D.

onekk, 2019-02-28:
Sorry for the late answer, the problem is related in writing a plugin for bCNC or working with bCNC code, wath i wanto to achieve is:

1) execute some "operation" using bCNC internals, like recording the working position at a certain points and saving them to a file.
2) elaborate this data maybe with and external program
3) load the resulting file in bCNC editor.

4) press run to execute the loaded file.

I don't achieve the goal to load the resulting file in the editor.

TIA and Regards.

Carlo D.

Harvie, 2019-02-28:
So you want to write bCNC plugin that uses external program to postprocess g-code loaded in bCNC?

onekk, 2019-03-01:
Yes and No.
The plugin will record some positions, and them pass a proper command line to a program, that produce a Gcode file.

Then this Gcode file has to be loaded in the editor and executed.

I haven't found the proper code to empty the editor and load the new file.

I have done many test, but no luck, analysis of the codebase don't help me so much.

Some operation are more keen to be done by external program, giving them the proper data.

I suppose that will also help many other developer.

Regards

Carlo D.

Harvie, 2019-03-01:
Have you seen this? #929

onekk, 2019-03-01:
I can't figure out how it could be related?

Capturing the points coordinates, is a work as esy to read some internals variables.

`def memA(self):
        #print("State: ", CNC.vars["state"])
        if CNC.vars["state"] == "Idle":
            mAx = CNC.vars["wx"]
            mAy = CNC.vars["wy"]
            mAz = CNC.vars["wz"]

            memvar =  "%f  %f %f"%(mAx,mAy,mAz)
            Utils.addSection("Memory")
            Utils.setStr("Memory", "mem_a", memvar)
            self.event_generate("<<Status>>",
            data=(_("Set mem_A to X: %f Y: %f Z: %f"%(mAx,mAy,mAz))))
        else:
pass`

The main problem now is how to empty the editor and reload the new file into it.

Regards

Carlo D.