vlachoudis/bCNC #1156: Text: Possibility of generating text in several blocks?

MARIOBASZ, 2019-02-07:
I believe that an improvement for the later ecanizado (profile, pocket) when generating text, would be that instead of being a single block, they are subdivided into blocks by closed line, for the later work profile - pocket:
An example: In the exterior use profile out (and I used islands, pocket in the box and difference with some modification).
In the interiors (A; O; & ..) use in or pocket profile.
Someone with a desire to implement it?
[image]

MARIOBASZ, 2019-02-07:
I think the solution to be implemented is simple: I have placed "(---------- cut-here ----------)" and split does the rest (again thanks! @Harvie, I use your solutions a lot!)

Harvie, 2019-02-07:
@MARIOBASZ Is that dibond/alucobond? Didn't knew you're a lawyer&programmer, i don't usualy see people doing both :-)

MARIOBASZ, 2019-02-07:
Jajaja, no, @Harvie, decidedly not: my daughter is the lawyer ... although my instruction was somewhat eclectic: two years of engineering, two years of sociology, both unfinished, ... periods where I read a lot of literature ... and I'm in the real estate sector. I am not a programmer at all !, I rely on my creative, inventive part, to find solutions. It is my intention to be able to dedicate myself to this (I have designed a universal cruise control), I have some ideas for CNC
It's not alucobond, probably the definitive poster will be. It is a miniature for sample. Here is called metalex (I think it's high impact pvc), simil metal.
I think I named it, I did the pocket operation that is interrupted in the letters (islands) instead of going over. First I had to do the profile operation, these are the islands (bit compensation problem)  Hugs