vlachoudis/bCNC #1158: Error: 'protocol "Spy" not known'

HumDP, 2019-02-08:
Hello guys.
I've been migrating from UGS Platform to bCNC and so far i'm very happy with the experience.

I have been struggling with a bug that freezes the GUI while the machine is doing work (milling or probing). Anyway, my intention was trying to solve it myself before bothering people here by checking the log of the comms between my PC and the CNC. According to what've read here, the 'Spy' handler is the way to go.

I run bCNC as Admin and set my port as '**_spy://COM3?file=bCNC_log.txt&raw_**' but when i try to open the port it pops up a error message that says **'_invalid URL, protocol 'Spy" not known_'** 

Am i doing something wrong?

My specs:
Sender: bCNC 0.9.14.52 (.exe Build)
Grbl version: 0.9
Controller: Woodpecker
OS: Windows 7 Ultimate
PC: Intel Atom 1.3 GHz 1gb RAM

I appreciate in advance whatever help you can give me.

Harvie, 2019-02-13:
Can you try to install bCNC using pip? The .exe build is experimental right now...

Harvie, 2019-02-14:
This looks like wrong pyserial version. Probably old version is bundled in .exe, because i've built it on Windows XP, which are not supported by latest pyserial...
Also be sure to use "spy" rather than "Spy".