vlachoudis/bCNC #1160: Help to improve trochoidal path and possible new plugin

MARIOBASZ, 2019-02-09:
I need help.
I want to improve the trochoidal path plugin, adding a comment that works as a marker at the beginning and end of adaptive diameters in the corners. I think it would serve something like this:
When executing the profile operation, the comment "(entered)" is inserted. I could not find where it is generated.
Can someone help me? @Harvie ?.

Harvie, 2019-02-09:
in lib/bpath.py:

```python
class Segment:
....
                self._inside = []               # auxiliary variable for tab and island operations
```
you can see that there is list called "_inside" that is part of each segment of path. You can store your information there as long as you do not convert path back to g-code. i think this is the best way to do it. it's made just for these cases.

But that means you need to do the trochoidal operation in single step single plugin. like this:

1.) convert g-code to path
2.) prepare trochoids, mark adaptive overcuts in _inside
3.) do adaptive overcuts
4.) convert path back to g-code.

MARIOBASZ, 2019-02-09:
@Harvie, I can not understand your date:

But that means you need to do the trochoidal operation in single step single plugin. like this:

1.) convert g-code to path
2.) prepare trochoids, mark adaptive overcuts in _inside
3.) do adaptive overcuts
4.) convert path back to g-code.
(I do not know how to copy an appointment on Github either)

In fact, I have not understood in profile and in pocket where it generates the routes, and especially where it adds "(entered)", for example (I thank you if you can indicate me.
Regarding adaptive, in lib / bpath.py, trochovercut is responsible for this: it generates the distance, and I generated a warning with a very small movement: Dpolice and an opposite movement -Dpolice: Trochoid must read this as a warning.
There is a drawback if I move the code generated from place, (move xx yy): this alarm is lost by rounding.
This is what I want to replace, so that it works as a warning as it does "- cut here -" in join after split
Then in trocoidal I made:
1- Control segment returns: If segment is a return, it does not generate trochoids
2-If adaptive alarm: Adaptive diameters from trochodal to zero for that segment.
3- Disable adaptive alarm
4-Execute segment returns (returns in a straight line)
(I do not know if you were saying the same thing)
Another question: Do you know how to do to generate only the routes in the corners in overcut profile, (or what is the same in for trochprofile)?

MARIOBASZ, 2019-02-09:
I mean to be able to generate this
[image]

 instead of

[image]

Harvie, 2019-02-11:
I don't think it's good idea to implement this using g-code comments.
You should use `app.gcode.toPath()` and `app.gcode.fromPath()`. Once you have converted g-code topath, you can use "_inside" for each line/arc to mark stuff as needed.

MARIOBASZ, 2019-02-11:
@Harvie, it has worked correctly for me as it is now: that trocoidal detects a warning of "adaptive initiation" (the adaptive section is always a segment, and then it returns to the starting point). However, my warning is artificial: two very small movements (round trip)
Only this is what I want to modify: the alarm that indicates "adaptive initiation"