vlachoudis/bCNC #1161: Write timeout

agintz, 2019-02-10:
Hello,
yesterday I get a SerialTimeoutException: "Write timeout" that I never received before. Since impossible to connect to the smoothieboard. I made installation with "pip install bCnc". Note I can connect and speak to the smoothieboard with another terminal (Termite) without problem.
version: bCNC-0.9.14.53
Thank you

Harvie, 2019-02-11:
I never had this bug before. Do you have some Arduino on hand, so you can try if that happens also with GRBL or just Smoothieboard? What version of Windows do you use? Have you tried different versions of pyserial? What do you see in "Port:" inputbox before you hit connect?

agintz, 2019-02-14:
Hi, OS is Win10. It seems coming from PySerial/Os interaction. I encounter the same problem with the python console, whatever the Pyserial version: 3.0.1 or 3.4. When I launch the commands :

```
>>> import serial
>>> ser = serial.Serial('COM3') 
>>> print(ser.name)
>>> ser.write(b'hello')
```
I get the write timeout exception too.
Everything is correct: Controller is SMOOTHIE, baudrate is 115200 as in my Smoothie config file.
Thank you for trying to help me

FerrisSalpetrum, 2020-04-28:
I have been using bCNC quite happily for over a year, and recommending it on the Shapeoko forums.  However,  this serial timeout has shown up suddenly.  It makes it impossible for me to use.   Can you PLEASE fix it and drop an exe suitable for windows in the repository,  or I am going to have to abandon bCNC

Elkinews, 2022-01-05:
hello everyone, i have the same problem, when i execute the program it works normally , but after an aleatory time this error ocurrs, and the machine stop her normal work,
its the first time using bCNC on windows , previusly i was using UBUNTU and this problem had never happends,
if any found a solution , i appreciate it.

i'm using also PoP OS and the problem was the same, just with ubuntu never had this problem