vlachoudis/bCNC #1162: Profiles

sfinexer, 2019-02-11:
You need to add profiles for the GRBL machine settings.
Let me explain, the machine may carry different loads, and its acceleration parameters, or other restrictions may be different.
Also in the profile to the settings of GRBL, there are various restrictions on the speed of movement, and in case of G-code errors, choosing the desired profile, you can physically prevent the machine from crashing.

Example: Laser burning, here the speed of maximum movement is not so important. In the case of nichrome, the white speed should be higher, as well as the acceleration is higher than during standard operation.

Harvie, 2019-02-11:
I don't understand what you need. You have to set your GRBL according to your machine physics. bCNC is not concerned about this.

sfinexer, 2019-02-11:
Have the ability to load settings ($$) into it from a certain profile. This is convenient when you need to comprehensively change one group of settings to another and vice versa.

In particular, it is necessary to periodically change the order of 10 settings, it is convenient to save the settings to a profile, and then create and manage GRBL settings profiles.
In this case, we have a group of settings.

MARIOBASZ, 2019-02-11:
I do not know if I'm getting it right: Record and re-record on eeprom, thinking it's not the best alternative.
I ask: accelerations does not depend on rigidity and characteristic of the machine more than if you use laser / spindle / hot wire (ok, maybe by spindle weight?)?
What I thought at some point, while BCNC does not handle G2Core, or the next grbl handle jerk, is if you could implement the jerk control from BCNC, but I do not know if it is possible, I trust that BCNC can join with G2core (especially for take better trochoidal)

Harvie, 2019-02-11:
I don't see what the problem is. You should set your $$ settings to highest speed and acceleration that your machine is capable of without losing steps (and leave some safety margin). Then if you need slower movement, you should use G1 and set F feedrate.

closed 2019-02-12 by sfinexer