vlachoudis/bCNC #1163: box tool dz bug

spanner888, 2019-02-13:
Looks like a copy paste error in the box tool.
		if nx > 0:
			self.nx = int(nx)
		else:
			self.nx = int(round(self.dx/nx))

		if ny > 0:
			self.ny = int(ny)
		else:
			self.ny = int(round(self.dy/ny))

		if ny > 0:
			self.nz = int(nz)
... so the code has TWO if ny > 0: .. and no if nz > 0:

Should the last one on line 74 actually be "if nz > 0:" ??

Harvie, 2019-02-13:
Does this cause any problems?

sfinexer, 2019-02-19:
Yes this is a mistake.
Of course, if nz> 0:

Harvie, 2019-02-21:
But can you explain what unexpected behaviour this causes? I don't want to read the code just to figure out what the real world consequence is, since you already know it.

Harvie, 2019-02-26:
Should be fixed by #1170

closed 2019-02-26 by Harvie