vlachoudis/bCNC #1167: Version 0.9.10 or 0.9.11

Andy-ABTec, 2019-02-18:
Apologies if I'm being thick here...

Just downloaded 0.9.11 as the "Latest release" but when I run it I'm seeing: -

[image]

Really confused now!

Harvie, 2019-02-18:
Where did you downloaded that from? Maybe try latest git master, there's lot of new stuff and we are heading to release it soon anyway...

Andy-ABTec, 2019-02-18:
Got it from https://github.com/vlachoudis/bCNC/releases/latest - which is why I was really confused...

I'll try the master

Harvie, 2019-02-19:
Anyway... I think this update check mechanism can be replaced in future using `pip install --upgrade` command. You can already use it when installing bCNC using `pip`. #1130

Andy-ABTec, 2019-02-19:
OK Thanks...

closed 2019-02-19 by Andy-ABTec