vlachoudis/bCNC #1169: Autolevel does not use rapid moves

FrenkK, 2019-02-21:
In 0.9.14, when I do grid levelling, all the moves are performed with the same low probe speed. This makes the levelling take longer than it needs to.
The program should use G0 Xx Yy Fsss for x-y direction moves after each probe. As speed value, the "Fast Probe Feed" value in the probing menu should be used. This would save a lot of time and would be more intutive.
A big thank you to everybody working on this awesome project! :)

Harvie, 2019-02-21:
Can you please provide pull request for this one? Sounds easy to fix...

FrenkK, 2019-02-21:
I'm a newbie at git and Python, but I'll try.

closed 2019-02-21 by FrenkK

Harvie, 2019-02-21:
> I'm a newbie at git and Python

I also was when i first started contributing to bCNC few months ago :-) We need more developers :)

FrenkK, 2019-02-21:
Ok, this took way too much time, but at least it seems to work. I added the pull request and I now even have a basic idea how Github works :)