vlachoudis/bCNC #1173: Error in code check mode.

sfinexer, 2019-02-27:
Error in code check mode.
I am using the latest version of GRBL, and the latest version of b-CNC.
In the G-code code according to the documentation, GRBL transferred it to the code check mode. I include the command $C at the beginning of G-code.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#g-code-error-handling
In this mode, GRBL does the same as :
https://github.com/vlachoudis/bCNC/pull/1171
But by means of GRBL.
I assume this behavior is due to an error:
https://github.com/vlachoudis/bCNC/issues/1108
Is there a video:
http://cnc.tiven.ru/ssr-2019-02-27_19.36.56.mp4
G-cod:
[test.cnc.zip](https://github.com/vlachoudis/bCNC/files/2911317/test.cnc.zip)

sfinexer, 2019-02-27:
As you can see in the video, error №1 goes first.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages
1) G-code words consist of a letter and a value. Letter was not found.

In the video, I also showed a piece of code on which proizoshol failure, and it is quite correct, and is identical to the previous sections of G-code.
The appearance of lines about GRBL versions are also confused.
I make an assumption that at the moment there was a failure in the b-CNC program and she began sending commands to GRBL that are different from the cnc code loaded into it.

Harvie, 2019-02-28:
GRBL gives you error:22. According to manual:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages

> 22 | Feed rate has not yet been set or is undefined.

BTW you don't have to put $C to your g-code file. You can just click $C button in terminal tab:

[image]

Harvie, 2019-02-28:
Also your g-code once again contains unsupported semicolon comment format. We've been discussing this before. GRBL only supports comments in round brackets.

`;Đàíäîìèçàöèÿ đàâíà = 210`

sfinexer, 2019-02-28:
@Harvie 
1) I brought a G-cod that works great and creates a picture (without the $C mode).
The problem manifests itself only when this mode is enabled.
2) Look carefully at the error number №22 should be later than the error number №1
3) I demonstrate in the video a piece of code where the problem occurred, it is correct and working. I will say more when you restart the error occurs in the other method.

Conclusion: this is not a G-code error.

sfinexer, 2019-02-28:
> Also your g-code once again contains unsupported semicolon comment format. We've been discussing this before. GRBL only supports comments in round brackets.
> 
> `;Đàíäîìèçàöèÿ đàâíà = 210`

Well, without a sign ";"
Video and cnc code:
http://cnc.tiven.ru/ssr-2019-02-28_18.03.18.mp4
[test.cnc.zip](https://github.com/vlachoudis/bCNC/files/2915555/test.cnc.zip)

Harvie, 2019-02-28:
What if you try the same code with UGS?

sfinexer, 2019-02-28:
> What if you try the same code with UGS?

I think the problem is not in the code ...
It works in the mode without $C. I make them pictures !!!

Harvie, 2019-02-28:
Please try to run the same check using different g-code sender. The error is raised by GRBL, so i am not sure that this is actual bCNC problem.

sfinexer, 2019-02-28:
> Please try to run the same check using different g-code sender. The error is raised by GRBL, so i am not sure that this is actual bCNC problem.

I do not have different g-code sender........

Harvie, 2019-02-28:
> Can you recommend for Linux?

UGS platform

sfinexer, 2019-02-28:
I did not succeed (((
http://cnc.tiven.ru/ssr-2019-02-28_19.03.25.mp4

sfinexer, 2019-02-28:
I am at a loss, I do not know what to do.
https://winder.github.io/ugs_website/download/#nightly-builds
2.0 Platform (Beta) | Java 8+ | July 28, 2017   Do not run.

2.0 Classic (Beta) | Java 8+ | July 28, 2017
http://cnc.tiven.ru/ssr-2019-02-28_19.03.25.mp4

sfinexer, 2019-02-28:
> Also your g-code once again contains unsupported semicolon comment format. We've been discussing this before. GRBL only supports comments in round brackets.
> 
> `;Đàíäîìèçàöèÿ đàâíà = 210`

https://github.com/gnea/grbl/blob/master/grbl/protocol.c
line 131:
`else if (c == ';') {
            // NOTE: ';' comment to EOL is a LinuxCNC definition. Not NIST.
            line_flags |= LINE_FLAG_COMMENT_SEMICOLON;`

sfinexer, 2019-02-28:
@Harvie 
Sometimes it is possible to get this error using the standard b-CNC tools.
Error received at the very end of the video.
http://cnc.tiven.ru/ssr-2019-02-28_22.26.04.mp4

Harvie, 2019-02-28:
> 2.0 Classic (Beta) | Java 8+ | July 28, 2017

In that video you only have 3 lines of g-code loaded... Probably you opened wrong file or something, because your code has more than 3 lines...

sfinexer, 2019-03-01:
> > 2.0 Classic (Beta) | Java 8+ | July 28, 2017
> 
> In that video you only have 3 lines of g-code loaded... Probably you opened wrong file or something, because your code has more than 3 lines...

The file is correct ...

sfinexer, 2019-03-01:
@Harvie 
I downloaded the latest UniversalGcodeSender problem left.
Video:
http://cnc.tiven.ru/ssr-2019-03-01_08.53.12.mp4

Harvie, 2019-03-01:
Perhaps there is some problem with your g-code. I've been using both UGS and UGS platform and never had issue...

sfinexer, 2019-03-01:
@Harvie can you provide your code?
I want to check if the error on your code repeats, then I have a problem.

sfinexer, 2019-03-01:
https://github.com/vlachoudis/bCNC/issues/1180#issuecomment-468746987
`
I've had not very big problem excluding those related to EMI and the CH340G on my Arduino that with some big gcode file seems to introduce some bug "forgetting some characters".`

This is probably related.
Since G-code has less than 10k lines, the error is rare.

onekk, 2019-03-03:
Some problem similar to those occurs to me when I launcha big gcode file, (I have not see the numner of lines of Gcode) but if the file is "big" and i split it in smaller file with different operations, i.e splitting the parts to be produced in different files, the machining  occurs correctly without problems.

Generally the error occur in a middle of some gcode, with grbl reporting a wrong line in the code that is correct in the gcode file and with "one" or some characters missing.

My board has the CH340G as a USB controller.

Hope it will help to find the problems, i have read some time ago but I'm not sure where that it occurs around 11K or 12k characters sent across the interface, maybe it was on some arduino forum.

Regards

Carlo D.

Harvie, 2019-03-03:
@sfinexer what USB-serial chip do you use? I already saw chinese chips doing funky stuff at 115200 baudrate... as @onekk reports.

sfinexer, 2019-03-04:
@onekk @Harvie 
My system, first writes the data to the SD card, and then sends it to the Arduino.
I'll have to wait a bit, I don't have a card reader to read the SD card, I ordered it, but the package is still on the way.
I see that the G-cod is not completely executed as it should, when the parcel arrives, I will say what the controller receives.
In other words, I made a buffer between b-CNC and GRBL. The system is working. You can turn off the desktop computer, the G-cod is running. However, due to the fact that I do not have a card reader, I cannot read the SD card, and find out exactly what the arduino-buffer is receiving.

onekk, 2019-03-11:
Maybe if you make talk the two arduino using the serial comunication (and not the USB)  I think the problem should not arise, as the serial line is not affected by the problem, it is related to the USB -> serial "bridge chip" that in the original Arduino Uno R3 is a 16u2 (If i didn't go wrong with the number, but it is an ATmel processor quite powerful with a USB com port ) and the clones use generally a more cheap CH340G (or maybe even a CH341G) that have the "lost character" problem.

Regards

sfinexer, 2019-03-11:
> Maybe if you make talk the two arduino using the serial comunication (and not the USB) I think the problem should not arise, as the serial line is not affected by the problem, it is related to the [...]

Yes ... I did not even think about it.
I also think that a bug in the CH340G or something similar.
I paid attention yesterday to the fact that one of the (others did not look) Chinese Arduino UNO on the CH340G chip (which should be there) does not have any marking.

Harvie, 2019-03-11:
I've been running my test on CH341 and it really does loose one byte from time to time. I've been testing 32B strings on baudrate of 2000000 and both of CH341 arduinos i've tested have lost byte like once every 2 MiBs transmited or so. Before i tested 64B strings on baudrate of 115200 and it seemed OK. But might not be perfect when combined with EMI from CNC machine...

sfinexer, 2019-03-11:
@Harvie And in what place is the loss of the character? at the beginning or at the end of the line?

Harvie, 2019-03-11:
There are no lines in test. Just infinite stream of bytes. Sometimes one gets lost.

sfinexer, 2019-03-11:
On the Chinese Mega2560 is the chip CH340G. I will look at the other one that is used for GRBL, it’s still too cold on the balcony (I have +11 degrees Celsius there)

Harvie, 2019-03-11:
Example of lost byte:

```
[harvie@anemophobia arduino-loopback]$ ./test-loopback.py /dev/ttyUSB3 
Total: 32.817566 MiB	Elapsed: 1681.163667 seconds      
ERROR!
Expected: 8feaafb3f46eebb681860d23d1df7ce8fd5b151c8548c14cc717faf65cbec751
Received: 8f  afb3f46eebb681860d23d1df7ce8fd5b151c8548c14cc717faf65cbec751
```

sfinexer, 2019-06-17:
After half a year, the card reader arrived ....
I can now state that the problem with data loss is that lines are lost during transmission to the Arduino. In the process of executing errors no!
The error is probably in CH340G or somewhere else.
But I declare that it is the loss of the STRINGS entirely, and not their parts.

Harvie, 2019-06-29:
Can you try with arduino that uses FTDI chip instead of CH340G ? These are slightly more expensive, but much more reliable.

Also for less than $10 you can buy chinese clone of saleae logic analyzer, 8channels 24MHz. It's very usefull for debugging. Looks like this:

[image]

You can then use pulseview (sigrok based) software on Linux to see what is going on in your wires and even decode serial traffic to see data in your serial line:

[image]