vlachoudis/bCNC #1176: How do not display verbose output on the Terminal

lab6-ifma, 2019-02-28:
Hello everybody.
I'm use the version 0.9.14.53 of bCNC and when I set the status report to $ 10 = 3 (to return the status and position of the machine), the terminal becomes very polluted with these values in sequence and I can not stop them. Below is a GIF for better understanding.

[image]

In the UniversalCodeSender has a check box called "**show verbose output**", which when unchecked omits GRBL's status report messages. In the bCNC tem alguma opção assim? Below a image showing it. When you uncheck these messages, they disappear.

[image]

Thanks everybody.

Harvie, 2019-02-28:
Also it's strange that machine reports "Alarm", but bCNC says "Connected". Is the alarm ignored with all `$10` settings?

lab6-ifma, 2019-02-28:
I was curious about what you said @Harvie, I do not know what happened, I just took the CNC to do some tests. Anyway I cleaned a ROM and did a new upload of the GRBL and it seems that solved. Thank you.

But also i have seen that this problem resembles the questioning #561, where @jpagliaccio say something like.

In my case, the terminal only returns the detailed output when the $ 10 = 3, however, when the $ 10 has different value, shows that it is not connected and the coordinates are no longer updated.

My goal is to disappear with this detailed output, but that the coordinates be displayed (as well as the state).

Other GIF below

[image]

Harvie, 2019-02-28:
> Anyway I cleaned a ROM and did a new upload of the GRBL and it seems that solved.

Yes, currently we only support latest GRBL. I guess that we might make parser more robust, so it will work with various GRBL setups... Patches are welcome :)

lab6-ifma, 2019-03-01:
Thanks for the comment @Harvie.

I use GRBL 0.9j because I started to plan a project about two years ago, but this project did not have continuous development, with large intervals of some months, then I did not update the GRBL, I thought it would still be stable.

However, due to your comment I tested the GRBL 1.1 here in the bCNC and really, the detailed output is no longer being displayed. I think I'm going to make some changes to my material to adapt it.

I think that a warning about unsupported GRBL version may be useful too, yeah.

Maybe if I have time I'll get some solution in the old GRBL0.py version, but I'm not a professional programmer... so...

Anyway, thank you @Harvie!

Harvie, 2019-03-01:
> I thought it would still be stable.

I guess that GRBL 0.9 is still stable, but it's interface is little bit different and currently bCNC project has nobody to maintain GRBL 0.9 support codebase.